dmroeder/pylogix #86: Writing large data causes 'Connection lost'

ramezanifar, 2019-11-01:
Hello,
Thank you for this great library. 
We are trying to replace the pycomm with pylogix because pylogix supports large connection size  (4000 bytes)
If I attempt to read even a larger array (>4000 bytes), it will fragment the data and reads it in multiple attempts successfully.
However,  the "write" method does not succeed and fails with error 'Connection lost'.
As I see in the source code for reading there are two associated functions _addReadIOI and _addPartialReadIOI but for writing there is one _addWriteIOI. 
Would it be possible to have the same functionality for Write function?
Thank you in advance

dmroeder, 2019-11-01:
I have not implemented the fragmented write, although I am aware that it exists.  I'll look into it again and see what I can do.  I'm a little busy at the moment so it might be next week.

dmroeder, 2019-11-06:
I have a solution that should work, however I think writing boolean arrays (my nemesis) might be currently broken so I have to fix that first.

Rather than implement fragmented write, which requires keeping track of byte offsets and such, I'll generate multiple writes depending on the connection size, amount of data being written and the tag name length.  I've tested it with multiple data types, array lengths and connection sizes, I just need to get those dastardly boolean arrays working. 

I'll probably push the changes to a test branch first to get your feedback.

dmroeder, 2019-11-21:
Still haven't forgotten, I've just been slammed with work.

dmroeder, 2019-12-17:
Okay, I _finally_ got to this one.  I ended up implementing fragmented write.  You should now be able to write arrays that don't fit into one packet.  Give it a shot and let me know what you think

closed 2019-12-21 by dmroeder

ramezanifar, 2020-01-22:
Thank you for the implementation.
With my CompactLogix L33ER controller, I did some tests. Here is the statistics of my experiment:

Read and write an array of floats of length 10
Reading stats (ms): average 2.1 | min 1.7 | max 8.6 
Writing stats (ms):  average 2.0 | min 1.7 | max 6.9
---------------------------------------------------------
Read and write an array of floats of length 100
Reading stats (ms): average 2.3 | min 1.9 | max 13.6 
Writing stats (ms):  average 2.4 | min 2.1 | max 6.0
---------------------------------------------------------
Read and write an array of floats of length 1000
Reading stats (ms): average 7.1 | min 5.5 | max 22.0 
Writing stats (ms):  average 9.0 | min 7.8 | max 19.3
-----------------------------------------------------------
Read and write an array of floats of length 2000
Reading stats (ms): average 11.4 | min 10.0 | max 50.0 
Writing stats (ms):  average 16.0 | min 14.5 | max 27.0
------------------------------------------------------------

Question:
Write takes longer, is there a reason and is it expected?

Thank you

dmroeder, 2020-01-23:
Write request packets are slightly larger than read requests.  So as you exceed one packet, writes will take increasingly longer compared to reads.  For example, reading 200 REAL's will take 2 packets, writing them will spill into a third packet.