dmroeder/pylogix #87: How to connect to Micro 820 processor?

mreese133, 2019-11-07:
According to your readme, I should only have to change the "micro800 = False" to True and it should work, but when I go to read a tag from my micro 820 it says Forward Open Failed. How do I correct this? I know it works with the larger models like compact and control logix processors, but I need it to work with micros as well.

dmroeder, 2019-11-07:
You are correct, you just need to set .Micro800 = True.  Make sure you are able to ping the PLC from the computer running the code, make sure you are reading/writing global variables only, you cannot read/write local variables.

Also, make sure you are using the latest.  A while back, I made some changes to the connection size that broke Micro8xx connections.

mreese133, 2019-11-07:
> You are correct, you just need to set .Micro800 = True. Make sure you are able to ping the PLC from the computer running the code, make sure you are reading/writing global variables only, you cannot [...]

I can ping the PLC from my computer, I can even use the Discover() function to find data on it, but I can't read or write to tags because it won't open a connection for reading and writing. It is a global variable, and it is a bit tag so there should be no issue with packet size. But when I try to open a connection it says "Forward Open Failed".

mreese133, 2019-11-07:
This is all with the most up to date version of your library.

dmroeder, 2019-11-07:
Hmm, what happens if you set the connection size to 500?

`comm.ConnectionSize = 500`

mreese133, 2019-11-07:
Same issue.

dmroeder, 2019-11-07:
Do you mind sharing your code with me?  You are welcome to email it to me if that is easier.

I don't have my Micro820 at the moment, however prior to loaning it out, everything was working.  Strange.

mreese133, 2019-11-07:
I am using your example code "log_to_csv.py" with a test tag I created. I set the micro800 flag to True in the eip.py. Do I have to reinstall your library after making these changes? I noticed that the current install creates encrypted python files, so I don't know which of those is actually being ran.

mreese133, 2019-11-07:
yep, that was the issue. I had to reinstall the library after making the change.

closed 2019-11-07 by mreese133

dmroeder, 2019-11-07:
Don't edit the eip.py file.  You would set the the flag in your script (the csv example).

```
import csv
from pylogix import PLC
import time

with PLC() as comm:
    comm.IPAddress = '192.168.1.9'
    comm.Micro800 = True
     
    with open('31_log.csv', 'w') as csv_file:
        csv_file = csv.writer(csv_file, delimiter=',', quotechar='/', quoting=csv.QUOTE_MINIMAL)
        for i in range(10):
            ret = comm.Read('LargeArray[5]')
            csv_file.writerow([ret.Value])
            time.sleep(1)
```

dmroeder, 2019-11-07:
Just a friendly comment, this is why I ask for code (which I rarely get).  Had you sent me the code, I would have seen that you were not in fact setting the Micro800 flag, that would have pinpointed the issue faster.  However, you figured it out :).  Glad you got it working!

So just keep in mind, you should never need to make changes to eip.py to use it.  You create an instance of PLC(), everything you need is done through that instance.

kodaman2, 2019-11-08:
@mreese133 this worked because when you do a pip install pylogix gets installed under a certain directory under the python root dir.

This is the wrong approach, if you try to connect to a clx now it won't work. So I would revert that and fix your script.

kodaman2, 2019-11-08:
>yep, that was the issue. I had to reinstall the library after making the change.

@mreese133 this worked because when you do a pip install pylogix gets installed under a certain directory under the python root dir.

This is the wrong approach, if you try to connect to a clx now it won't work. So I would revert that and fix your script.