dmroeder/pylogix #88: Read tag Descriptions?

spadgenske, 2019-11-11:
I am looking for a way to read the tag descriptions along with the name and value, is this theoretically possible? I know I can parse a tag output file or a .l5x file but reading the descriptions on the fly would be super handy.

dmroeder, 2019-11-11:
I am unaware of any way to do this, however I'm sure it is technically possible.  Rockwell has never documented how descriptions are saved and accessed as far as I'm aware, it would require a lot of reverse engineering that I'm not sure I'm up for.

I've spent many hours trying to mimic exactly what RSLinx does, it doesn't always produce the same result as RSLinx does.  For example, accessing local variables on the Micro8xx controllers.

kodaman2, 2019-11-12:
I would be surprised if Rockwell is doing anything meaningful on the cpu
side with tag descriptions. They don't serve any purpose logic wise. I
think they are only saved in the memory card when you save the project to
nvram and on the acd file offline.

What you see in rslogix 5000 is most likely being read from the local acd,
probably same way how you would parse the xml.

This is exactly how things are done in Siemens plc's too if you took an
upload you'll get nothing but logic without tag names or comments all that
is stored in the project offline like the acd file.

I have thought of making helper functions and this can certainly be one of
them so users spend zero time learning how to parse an xml, but eventually
all these helper functions would be on a separate repo as pylogix should
stay close to its core.

So for example all you'll need is provide l5x, and tag name and you'll be
able to retrieve whatever properties are inside the tag name l5x.

evaldes2015, 2019-11-12:
I believe older versions of RSLogix work as you described. I believe newer version store the tag comments in the PLC. Starting with V21 and L6x the tag comments are stored in the PLC.

kodaman2, 2019-11-12:
@evaldes2015 Yes stored in nvram, however I don't think they are used with plc scan cycles. Comments are not in RAM nor working memory. However I could be wrong, this is just how I think it is implemented. Maybe there's documentation somewhere on the net on this. 

Remember how much memory strings use, seeing how comments don't impact logic is probably why is not implemented within cip transactions. Without this I don't see a way of retrieving comments, might as well get them from l5x since is essentially the same thing.

ottowayi, 2019-11-12:
I actually did some research trying to reverse engineer the extended properties in FTView.  I figured out it was a Write Fragment request with some byte constants and a property number, then the value was returned in the response.  I couldn't ever get my packets to line up exactly with what I was seeing from FTView, but I didn't have much time to invest into it.

dmroeder, 2019-11-12:
The comments are stored in the controller, the details of how/where, I don't know.  There are CIP commands to get them however they are undocumented, making them very difficult to use.  Not trying to plant a seed of hope here.

The few times I've went down one of these rabbit holes, it always appears that there are commands required prior to the command you are trying to mimic.  When you are looking at bytes in a sea of wireshark packets, it’s impossible to tell which ones are related.

Years ago, I was digging into the CIP command to change the controller mode.  I can send the command to a controller v19 or earlier, starting at v20, it rejects the command.  I can watch RSLinx send the command, however when I send it, it gets rejected.   Turns out, starting at v20, a full upload of the program has to be made, which sort of authenticates RSLinx, before the PLC will accept the change mode command.

ottowayi, 2019-11-12:
Yeah that was about where I got to, I had most of the request packet mapped out, but figured there had to be some sort of authentication happening before then.  If it was just a simple service, someone else would have reversed engineered it by now.

kodaman2, 2019-11-13:
@ottowayi I would guess this is purely by design for security. If you can read the comments and they are well documented you could reverse engineer some rungs, granted it would take a lot of work.

ottowayi, 2019-11-13:
I don't know, that's giving Rockwell too much credit.  I can already read the tag list, read any tag value, write any tag value, etc.  I don't think reading descriptions adds much security.. it's just one feature they can use to sell you their HMI that competitors can't use.

closed 2019-12-21 by dmroeder