dmroeder/pylogix #89: AttributeError: 'Response' object has no attribute 'Value'

paneee, 2019-11-14:
I have a problem when using the library on linux. It works very well on windows. In linux (raspbian) when reading a tag:
```
Traceback (most recent call last):
  File "test.py", line 19, in <module>
    print(ret.Value)
AttributeError: 'Response' object has no attribute 'Value'
```
I attach a picture with a comparison:
[image]

I appreciate your work and thank you for your help

dmroeder, 2019-11-14:
The issue isn't with Raspbian, I do 100% of my work with this in Linux.  I have a Pi at my desk running it too.  Why run it with sudo?

From raspbian, can you ping the PLC?
From your raspbian VSCode terminal:

```
python3
import pylogix
pylogix.__version__
```

paneee, 2019-11-14:
[image]

paneee, 2019-11-14:
After changing the letter 'V' to the lowercase 'v' in the line:
`print(ret.Value)`
change to:
`print(ret.value)`
The problem is solved.

I don't know why but it works :)
Thank you for your reply and your time.

dmroeder, 2019-11-14:
Wait a sec, 0.5.1?

It seems that you installed a version at a moment of chaos in this project.  There was a minute when I changed how data was returned, using the Response class, where the responses were lower case.  I think this was the more pythonic way to do it, however the rest of the project was "PascalCase", so I flip-flopped for consistency.  I'm not sure where  0.5.1 came from though, I'm at v0.4.12.  I know here were some pull requests accepted, maybe that version was pulled in, then fixed.

I'd bet that if you checked the version between raspbian and the other machine you were testing it on, they'd be different.

Anyway, sorry for the confusion, I'd recommend uninstalling pylogix and reinstalling it using the instructions in the readme.  Your version should be 0.4.12 and you will use print(ret.Value).

kodaman2, 2019-11-14:
0.5.1 is the PR #65, how in the world did you manage to installed that? it was never merged into master. I bet he installed it straight from the owner of PR #65 since he changed the pip install to point to his.

dmroeder, 2019-11-14:
Ah good call, I knew it was around that point, I was scratching my head a bit...

kodaman2, 2019-11-14:
Yeah it wouldn't have happen normally but he change the readme to point to his repo. I requested him to fix that long time ago looks like it might be a dead PR. So someone searching for pylogix can stumble across that rogue repo lol.

@paneee in the future please look at how many stars the project has, usually the original has more, and forks a whole lot less.

paneee, 2019-11-14:
Thank you for the detailed explanation and time because I was lost in the subject.
I didn't know that the pip version is not your last one.
For the future how to check what version is in pip / pip3?

kodaman2, 2019-11-14:
Glad to help, no worries. Just as Dustin pointed out:

```
pylogix.__version__
```

and then if you go to [releases](https://github.com/dmroeder/pylogix/releases) you can see latest at the top.

paneee, 2019-11-14:
Ok, Thanks

closed 2019-11-14 by paneee