dmroeder/pylogix #90: What is the proper way to read string tags?

e2ibrobbins, 2019-12-04:
I don't see an example show how to do it.  I see one for writing to them.  I have a program called "VS38001" with tag called Unit that is a UDT with an element Batch with an element ID.  Unit.Batch.ID is a string.

I tried 
    ret = comm.Read('Program:VS38001.Unit.Batch.ID')
    print(ret.Value)

And many variations including ...DATA[0] and the like.

e2ibrobbins, 2019-12-04:
I think I am answering my own question.  Reading other controller-level and program-level string tags seems to work, but the limitation seems to be handling of UDTs.  This seems confirmed by other issues.

dmroeder, 2019-12-04:
If the tag "ID" is a standard STRING, you have it right and it should return the value of the string.  Reading a STRING is no different than reading a DINT, for example.  The fact that the STRING is part of a UDT doesn't matter.

If the tag "ID" is a custom defined string, you will not be able to read it without doing some work yourself.  

Can you confirm this?

e2ibrobbins, 2019-12-04:
Thanks for the prompt reply.  Yes, the "ID" tag is a standard string.  The error was found.  The structure of the UDT was slightly different than I was first told.  Examining it with the PLC programmer revealed that the tag was "Unit.BatchID" not "Unit.Batch.ID".  Knowing that, I can read the tag.  This issue is mine, not the pylogix library.

closed 2019-12-04 by e2ibrobbins

dmroeder, 2019-12-04:
Ah okay, nice!  Thanks for the update.