dmroeder/pylogix #91: reading multidimensional array

gnail13, 2019-12-05:
Hi all,

I have a project that I need to retrive a two-dimensional array size of [1000][7].  How can I read it?  Thanks in advance.

Lan

dmroeder, 2019-12-05:
You use the same syntax as in RSLogix/Logix Designer:

MyTag[0,0]

gnail13, 2019-12-05:
Can I use indirect addressing like MyTag[i,j] to access the array other than tag.Read(MyTag[0,0], 100)?

gnail13, 2019-12-05:
In other words can I traves the array with for loop?  So I can read the data on first dimension MyTag[i,0].

dmroeder, 2019-12-05:
You could do something like:

```
for i in range(10):
    tag_name = 'MyTag[{},0]'.format(i)
    ret = comm.Read(tag, 100)
```

Is that what you were thinking?

gnail13, 2019-12-05:
I am not sure that's the result I wants but I would try it.  Thank you.

dmroeder, 2019-12-05:
First iteration would get you 100 values starting at Mytag[0,0] (MyTag[0,0-99])
Next iteration would get you 100 values starting at Mytag[1,0] (MyTag[1,0-99])
and so on

gnail13, 2019-12-05:
Thank you dmroeder.  That's what I need.

closed 2019-12-06 by dmroeder

AlePanto, 2023-03-24:
I am having similar issues:

Need to read the entire TD_Array DINT[10,7] 

'''What is wrong with the code?'''

from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = '192.168.1.15'
    ret = comm.Read('TD_Array[10,7]' )             #what do I nee to specify?
    print(ret.Value)

'''Thank you for the help'''

dmroeder, 2023-03-24:
Install the latest version (0.8.8).  I broke something with multi dimension arrays.

AlePanto, 2023-03-24:
I upgraded before asking on the forum! 
Well... I hope you and community can solve it. Thank you!!!

AlePanto, 2023-03-24:
Nevermind!!!!
Iit works, Thank YOU!!!!!!!