dmroeder/pylogix #94: Concurrent connections on L33ER

gideoncaringal, 2019-12-13:
Hi there!

I'm working on reading tags from an L33ER. I've been able to read/write tags using the examples given.  

Next test is to use python to read data from an M-Bus energy meter and write these into PLC tags. This would happen maybe every 2 or 5 minutes to conserve battery life on the meter. 

I would like to get some tips on how to implement a single program that reads a number of tags every 5 seconds, and writes one tag every 5 minutes. As I understand, I cannot open a connection that reads several every 5 seconds, and then open another connection every 5 minutes to write a tag. So my first thought is to just use datetime and some if statements and just keep the program looping. 

Is there a better/smarter way to do this?

Thank you in advance. I'd appreciate any sort of help on this.

dmroeder, 2019-12-13:
There are a few ways to skin this cat, one simple way is to is the time module to create the 5 second delay.  After 60 reads, then write.  No need to have multiple connections, something like this:

```

import time
from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = '192.168.1.10'
    read_count = 0
    loop = True
    while loop:
        try:
            r = comm.Read('tag1').Value
            time.sleep(5)
            read_count += 1

            if read_count >= 60:
                w = comm.Write('tag2', 123)
                read_count = 0
        except KeyboardInterrupt:
            loop = False
```

gideoncaringal, 2019-12-18:
That's a great example and a big help into adding more functionality into my loop. Thank you!

closed 2019-12-21 by dmroeder