dmroeder/pylogix #96: Write string to Micro800

bdamori, 2019-12-23:
I just downloaded latest version of Pylogix.  Some how writing string to Micro800 stopped working.  The version I loaded around first of Oct 2019 worked great.

dmroeder, 2019-12-23:
Thanks for the heads up, I'll take a look when I get back to a computer. It will be a few days.

dmroeder, 2019-12-23:
In the meantime, can you post the error you get?

kodaman2, 2019-12-23:
Also please confirm version:

```
python import pylogix pylogix.__version__
```

Also confirm `Micro800` is set to `True` on your script not on `eip.py`. I've seen some users change that value on eip.py and since you mentioned you installed it that could very well be the issue.

bdamori, 2019-12-23:
Sorry I copied the old pylogix file from old PC to new PC install and got it to work . I believe I remember the error was on line 288 of the eip.py file.  I think a divide  by zero error .

bdamori, 2019-12-23:
I did have Micro800 set to true. If I get time to reinstall newer version I’ll get you the version number and actual error. It was the newest version as of yesterday or the day before.

bdamori, 2019-12-23:
The version I’m using that works is
__version_info__ = (0, 5, 1) I downloaded this around Oct 1 2019
When I look at what is on GITHUB now it is __version_info__ = (0, 5, 0)
I believe that this version  does not work.  Like I said I’ll try and retest next week.

kodaman2, 2019-12-23:
Thanks for the info. If the error is at line 288 then the bug could be line 1542 since you mentioned a divide between by zero error.

If you have time to confirm with a screenshot that would be great. In the readme you can see how to install with pip by versions. Just go on releases and go down the line until strings work.

kodaman2, 2019-12-23:
Version 0.5.1 is not from this repo it was a fork and that guy changed the version but his code hasn't been merged, @dmroeder we really need to do something about that rogue repo, either way use the releases from this repo and install with pip as mentioned previously. My guess is that 0.4.13 or 12 will work fine.

bdamori, 2019-12-23:
Thanks I’ll try one of those version .

bdamori, 2019-12-27:
HI,
Neither   0.5.0 or 0.4.13 versions worked for me. Version 0.4.12 did work fine.

Below is the errors from the 2 versions that did not work.
Traceback (most recent call last):
  File "/home/linaro/Desktop/micro.py", line 8, in <module>
    comm.Write('Outbound','Have a Mark')
  File "/home/linaro/.local/lib/python3.7/site-packages/pylogix/eip.py", line 104, in Write
    return self._writeTag(tag, value, datatype)
  File "/home/linaro/.local/lib/python3.7/site-packages/pylogix/eip.py", line 291, in _writeTag
    writeData = self._convert_write_data(b, dataType, writeData)
  File "/home/linaro/.local/lib/python3.7/site-packages/pylogix/eip.py", line 1498, in _convert_write_data
    limit = int(space_for_payload / bytes_per_value)
ZeroDivisionError: division by zero

dmroeder, 2019-12-27:
Interesting, I'll take a look.  Can you copy/paste the exact print out of the error?  Unfortunately, I no longer have my Micro820 PLC, or I would test it myself.

bdamori, 2019-12-27:
I did past the errors on last email

bdamori, 2019-12-27:
Let me know if/when you get newer versions fixed . I’ll try and run them.

Thanks
Bernie

dmroeder, 2019-12-27:
Haha, sorry, I'm having a hard time today apparently :)

kodaman2, 2019-12-27:
> Haha, sorry, I'm having a hard time today apparently :)

Tbh is an odd format when you reply from emails lol. I'm having hard time reading the responses too.

dmroeder, 2019-12-27:
I know what the problem is.  I'll email you something to test if you don't mind...

bdamori, 2019-12-27:
Sure that’s fine

bdamori, 2019-12-29:
[eip.py_line1542.txt](https://github.com/dmroeder/pylogix/files/4007869/eip.py_line1542.txt)
I made a change to eip.py  on line 1542. I'm new to Python so this may not be the correct change but it worked.see attachment

dmroeder, 2020-01-03:
I take it that the change I emailed you didn't work?

bdamori, 2020-01-03:
I did not get/find your email (dmroeder).  . I thought I saw one and when I went back  the next day I could not find it. If you send it again I will test it.

dmroeder, 2020-01-03:
I sent it again, hopefully it's not going into your spam.

bdamori, 2020-01-03:
Thanks, Changing line 71 in eip.py fixes the issue

dmroeder, 2020-01-04:
I finally got a moment to push the change.  That should take care of this one.  Thanks for your help and testing!