dmroeder/pylogix #101: Issue with reading multiple tags from Micro820 PLC

JustBru00, 2020-01-17:
Code:
```Python
import sys
from pylogix import PLC

if len(sys.argv) == 1 or len(sys.argv) > 6:
    raise ValueError('Provide first an ip address then up to 5 tag names seperated by spaces.')

tag_list = ['R1_AIR_TEMP','R2_AIR_TEMP','R1_PROD_TEMP','R2_PROD_TEMP']

with PLC() as comm:
    comm.IPAddress = sys.argv[1]
    comm.Micro800 = True
    print(sys.argv[2:len(sys.argv)])
    '''
    ret = comm.Read(sys.argv[2:len(sys.argv)])
    '''
    ret = comm.Read(tag_list)
    for r in ret:
        print(r.TagName, '|', r.Value, '|', r,Status)
```

I am trying to read from a list of tags following your example (05_read_multiple_tags.py). I am receiving the following error message:
```
~python3 read_multiple_tags.py 192.168.0.163 R1_AIR_TEMP R2_AIR_TEMP
['R1_AIR_TEMP', 'R2_AIR_TEMP']
Traceback (most recent call last):
  File "read_multiple_tags.py", line 16, in <module>
    ret = comm.Read(tag_list)
  File "/home/cc/.local/lib/python3.6/site-packages/pylogix-0.6.0-py3.6.egg/pylogix/eip.py",
  File "/home/cc/.local/lib/python3.6/site-packages/pylogix-0.6.0-py3.6.egg/pylogix/eip.py",
  File "/home/cc/.local/lib/python3.6/site-packages/pylogix-0.6.0-py3.6.egg/pylogix/eip.py",
struct.error: unpack_from requires a buffer of at least 2 bytes
```

I am using pylogix version: 0.6.0
```
Python 3.6.9 (default, Nov  7 2019, 10:44:02)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylogix
>>> pylogix.__version__
'0.6.0'
>>> exit()
```

Python version: 3.6.9
```
>>> import sys
>>> sys.version
'3.6.9 (default, Nov  7 2019, 10:44:02) \n[GCC 8.3.0]'
>>>
```
I am probably just missing something basic as I haven't messed much with python, but I can't figure out if this is an issue with my code or a compatibility issue with pylogix and my version of python.

Thanks for your time,
Justin B.

dmroeder, 2020-01-17:
Micro800 series doesn't support the multi service object, you have to read/write them individually.  Though, it shouldn't be crashing, it should be returning "service not supported" for the status. 

I'll fix the crashing once I get my hands on a Micro820 (mine is loaned out at the moment) but you still won't be able to do it.  Only the Logix platform supports multi-service object.

JustBru00, 2020-01-17:
Sounds good. I figured there was a simple explanation.

- Justin B.

closed 2020-01-17 by JustBru00

dmroeder, 2020-01-17:
Unfortunately those things are pretty limited.  I guess that can be expected for the price.