dmroeder/pylogix #102: Forward Open Failed for 1756-ENBT/A

AndrewStuhr, 2020-01-18:
Hello

Do you know of any issues with the ControlLogix 1756-ENBT/A?  Python returns "Forward Open Failed" for both .Read() and .GetTagList(), although it successfully grabs the PLC' ip address using .Discover().   .Read() and .GetTagList() are working for another ControlLogix model, 1768-ENBT/A.  

Maybe there's an ethernet configuration difference between the 2 PLCs.  Can you think of a setting I should compare between the 2 projects.

Thanks

dmroeder, 2020-01-18:
There is not issue that I'm aware of.  Can you show me your code?

AndrewStuhr, 2020-01-18:
#the following import is only necessary because eip.py is not in this directory
import sys
sys.path.append('..')
from pylogix import PLC

try:
	tagList = []
	with PLC() as comm:
		comm.IPAddress = '192.168.50.28'
		tags = comm.GetTagList()
		
		for t in tags.value:
			tagList.append([t.TagName, t.DataType])

	print(tagList)

except Exception as e:
	print(e)

AndrewStuhr, 2020-01-18:
[image]

AndrewStuhr, 2020-01-18:
Just changing the IP makes it work for other PLCs

dmroeder, 2020-01-18:
You need to also add comm.ProcessorSlot = 3

AndrewStuhr, 2020-01-18:
You're the fucking man

dmroeder, 2020-01-18:
Haha, thanks.  For more info, the default value is 0, so that is why it works with CompactLogix without specifying that parameter.  You need to specify it when the controller is in a different slot.

AndrewStuhr, 2020-01-18:
I'm interested in automating the slot number selected so it always selects the PLC.  Do you recommend looping over different slot numbers and checking the response, or is there a better way to determine the PLC's slot number?

dmroeder, 2020-01-18:
You could probably use GetModuleProperties() to figure out what device is in each slot.  Example 28 might help you along.  It will discover devices, then query modules as if everything was a 17 slot rack.

AndrewStuhr, 2020-01-18:
I'll check it out, thanks!

AndrewStuhr, 2020-01-18:
It seems that I must specify comm.ProcessorSlot before using GetModuleProprties().  I get "Forward Open Failed" if I leave the default value 0.  Is this expected?

kodaman2, 2020-01-18:
Did you run the whole 28 example? Can you post `network_audit.txt`?

You need to specify a slot for `GetModuleProperties` line 35 of example 28 loops through all slots tp figure out which slots are plc's.

Though you should be able to run `Discover` without a specifying a slot.

AndrewStuhr, 2020-01-18:
Ok I must have missed something, I'll retry that, thanks

dmroeder, 2020-01-18:
> It seems that I must specify comm.ProcessorSlot before using GetModuleProprties(). I get "Forward Open Failed" if I leave the default value 0. Is this expected?

That doesn't sound right to me.  I had to implement "unconnected send" for the purpose of not having to connect to a specific controller prior to sending this request.

If you feel like experimenting, find:

```
    def _getModuleProperties(self, slot):
        """
        Request the properties of a module in a particular
        slot.  Returns LgxDevice
        """
        if not self._connect():
            return None
```

comment out the last 2 lines (if statement and return).  See if that makes a difference.

dmroeder, 2020-01-20:
I must have not had enough beer this weekend.  Disregard my last post, you must connect to a PLC first to retrieve a modules properties.  It is the PLC that is responding, not the module itself.  So yes, you must specify the processor slot first.

AndrewStuhr, 2020-01-27:
Hi again

I'm looping over 17 slots and using GetModuleProperties() to determine which slots to try reading tags from.  However, there's a timing issue.  For our compactLogix PLC, this happens very fast for slots 0-2 & 10-16, but slow for slots 3-9 (about 4 seconds per slot).  As a result, it takes 40-45 seconds to scan each slot.  However, for controlLogix PLCs, there's no issue; it scans each slot lightning fast.

Any thoughts/suggestions for why GetModuleProperties() is slow for some of the compactLogix slots?

dmroeder, 2020-01-28:
Could you list the catalog numbers of all of your CompactLogix modules?

Although we use a lot of CompactLogix processors, we use very little in rack I/O

BTW, disregard my previous message, commenting out those lines isn't the problem.

dmroeder, 2020-02-20:
@AndrewStuhr , I pushed a change that should take care of requiring a controller to be in slot 0 for get module properties to work.  As for the speed issue, I'd still need some more details from you, I don't have the hardware to test, all of our ControlLogix racks are 4 slot and CompactLogix racks are typically 2 modules.  Most of our I/O is distributed over Ethernet.

AndrewStuhr, 2020-03-12:
Hey, I'm not able to recreate this issue.  We've had some PLCs swapped out at work, and I don't remember what the original PLC model was.  I'll let you know if I encounter it again.

closed 2020-03-21 by dmroeder