dmroeder/pylogix #103: Emerson

htrifol, 2020-01-21:


evaldes2015, 2020-01-21:
Correct me if I'm wrong, but I believe you're wanting to do implicit messaging which is used for I/O.
Pylogix supports explicit messaging. They are not the same thing.

htrifol, 2020-01-21:
Not really, it should be implicitly but I think the issue is the credentials issue that need to be sorted.
Getting passed the admin admin thing should be it.... I guess ;)

kodaman2, 2020-01-22:
@dmroeder can confirm but pylogix only reads tags from a plc that supports etherip as far as I know.

dmroeder, 2020-01-22:
I have no idea, I don't use Emerson products.  I doubt that you will get any thing useful with those devices using this library, it supports a couple of CIP objects specific to Rockwell PLC's. 

There are many CIP objects, a whole block of them are reserved as "vendor specific".  I would bet all Ethernet I/P devices support List Identity (what I called Discover) object, so it doesn't surprise me that your gateway and transmitter respond.  How your devices communicate to Rockwell PLC's is probably using the vendor specific objects, not the CIP Read and CIP Write objects.

All devices probably support Register session and Forward Open, where the connection parameters are established.  So sometimes it appears that pylogix will attempt to communicate to a device but it probably won't.  This is only intended to work with Rockwell PLC's.

closed 2020-01-27 by kodaman2