dmroeder/pylogix #104: Support for Eithernet/IP Class I  (implicit message)

ramezanifar, 2020-01-31:
Can this library be expanded to support EIP class 1? In order to be an adapter to mimic an IO for a PLC or mimic a scanner to read IO?
[OpENer](https://github.com/EIPStackGroup/OpENer) is a library in C. But there is nothing in python. It would be awesome if you consider this.
Thanks

dmroeder, 2020-02-01:
While I agree that it would be neat, I have no plans to add that functionality.  I have experimented with this idea though.  It proved not very reliable at sub 100ms RPI, the timing of the socket seemed too unpredictable.  

I won't say that it will never happen, I just have too many things to do and not enough time to do them.

kyle-github, 2020-02-01:
Have any notes on how to set up the ForwardOpen for implicit messaging?   I'd like to give it a try in my library.

dmroeder, 2020-02-01:
I sent you an email Kyle

kyle-github, 2020-02-01:
Thanks! I'll take  look.