dmroeder/pylogix #105: Multiple PLC comm

Quanho90, 2020-02-20:
What is the best way is getting tag values from multiple PLCs? Can we open comm with all of PLCs at the same time or just open comm then close 1 by 1?

Extra question: How do you know the way timer is packed in UDT? I cannot find this information in any Rockwell manual. The only thing I found is Timer includes PRE,ACC, etc.

Thanks

dmroeder, 2020-02-20:
I would create an instance of the driver for each PLC:

```
from pylogix import PLC

plc1 = PLC('192.168.1.10')
plc2 = PLC('192.168.1.100')
```
etc...

As for the timer, I figured out how the data was packed by reading one over and over with different values/bits, monitoring the changes in wireshark.  Figuring out an actual UDT is easier because you can see the structure in the L5K export.

ottowayi, 2020-02-21:
If you're interested in doing full UDT reading, you can checkout my code [here](https://github.com/ottowayi/pycomm3) for an example of it working.  Part of the initial startup is reading all the tag definitions from the controller, then using those it recursively parses thru the structure reply data to get every attribute.  I've noticed that the built in structs all seem to have this CTL attribute of type DWORD that is not shown in Logix.

th3dj, 2020-03-30:
I came up with a cleaver way to handle this in my project, instead of having to write the code for each plc try this.

```
ips = ['0.0.0.0', 'plc1'],['0.0.0.0', 'plc2'],['0.0.0.0', 'plc3'],['0.0.0.0', 'plc4']
def flag(x,loc):
    ref = x.Read('Error_Flag[0]',4)
    if ref.Value is None:
        print(ref.Status)
    else:
       #whatever you want to do with the output.
while True:
    for h in ips:
        PLCs.IPAddress = h[0]
        flag(PLCs, h[1])
```

So for me at least i was able to get it to cycle through the IP's and then the second part of my array allows me to know what plc i am working with if the flag goes true. That way I can make it do what i want depending on the set up.

dpatel2016, 2020-12-09:
How to connect multiple Micro800s in the same script? They each have different IPs

TheFern2, 2020-12-09:
> I would create an instance of the driver for each PLC:
>
> ```
> from pylogix import PLC
>
> plc1 = PLC('192.168.1.10')
> plc2 = PLC('192.168.1.100')
> ```
>
> etc...
>
> As for the timer, I figured out how the data [...]

Did you happen to try this? create an instance of the PLC object for each PLC.

dpatel2016, 2020-12-09:
Yep, it didn't work the first few time because was using ethernet switch with computer connected to it. I had two PLC, raspberry pi, and laptop connected to the switch. It worked once the computer was removed from the switch.

dpatel2016, 2020-12-13:
How do you ignore the plc that is not on while reading multiple plc in one script. I am reading 2 plc and sometimes one of them is off so it delays readings for both. Is there anyway to fix these?

TheFern2, 2020-12-13:
There's prob a few ways to handle that, for me the best way is to ping the plc https://github.com/kyan001/ping3 somewhere in the beginning of your loop to see if plc is online and if it is then proceed to connect to it with pylogix PLC object.

dpatel2016, 2020-12-13:
I tried that but it gives me permission error, which means I have to use sudo to run this script and it will require password at the startup.

TheFern2, 2020-12-13:
You prob need to run the script from root on startup, or use cronjob from root. Regardless this isn't a pylogix issue, please use SO for these types of questions.

dmroeder, 2020-12-13:
@dpatel2016 you could try setting SocketTimeout lower, say 1 second.  If that delay is too much, you'll probably want to look into threading.

dpatel2016, 2020-12-13:
I will try that, thank you!

closed 2021-05-10 by dmroeder