dmroeder/pylogix #106: Reading Custom String

DCogent, 2020-02-24:
Hi Dmroeder,
  As always I love this program.  I came across an issue today where when I try to read a custom 2 letter string I created, the data comes is goofy. It reads in like this:
"\xab\x11\x02\x00\x00\x00O1\x00\x00"

O1 is the actual string. How do I keep everything else from coming in. 

                                                                                         Thank You!

evaldes2015, 2020-02-24:
Hmm... One program I have returns a string, another returns something similar to what you are seeing.

The one that works reads a regular string tag. The one that returns the odd result is part of a UDT.

Is the tag you are reading a UDT?

evaldes2015, 2020-02-24:
You can always read the string LEN then read the string DATA. That will work.

DCogent, 2020-02-24:
I am reading from a custom String. Its only got a length of 2. 
How would i go about reading the length prior to the data?

evaldes2015, 2020-02-24:
<tag name>.LEN

evaldes2015, 2020-02-24:
`<tag name>.LEN`

dmroeder, 2020-02-24:
I wouldn't call it an issue per se, this is just the nature of how custom strings are handled.  They are no different than a UDT.  So since the data type is unknown, it returns the raw data for the user to parse.  One way would be to read the .LEN of the string to know how many characters are in it, then read the appropriate number values from .DATA

```
from pylogix import PLC

with PLC('192.168.1.78') as comm:
    str_len  = comm.Read('CustomString.LEN').Value
    data = comm.Read('CustomString.DATA[0]', int(str_len)).Value
    value = ''.join([chr(d) for d in data])
    print(value)
```

DCogent, 2020-02-24:
Im reading it from a list of tags. 
Is there any way i can add it to the list of tags that the pylogix looks through

dmroeder, 2020-02-24:
I would probably write a function to read a custom string, using something similar to what I did above.  Then in your program, that is getting the tag list and doing the reading, if you see the data type is your custom string data type, call your custom string function instead of using .Read() directly.  Something like:

```
for tag in tags:
    if tag.DataType = "MyCustomString":
        value = my_custom_string_reader(tag.TagName)
    else:
        value = comm.Read(tag.TagName)
```

DCogent, 2020-02-24:
This is a short bit of where im reading the list. So i would just need to add it here if i read that Custom String
                                        ValueList = []
                                        for tag in taglist:
                                            value = comm.Read(tag)
                                            now = CurrentTime                 
                                            ValueList.append(value)

DCogent, 2020-02-24:
I think i get it.

dmroeder, 2020-02-24:
Maybe  something like this:

```
def read_custom_string(tag_name):
    str_len = comm.Read('{}.LEN'.format(tag_name)).Value
    data = comm.Read('{}.DATA[0]'.format(tag_name), int(str_len)).Value
    value = ''.join([chr(d) for d in data])
    return value
```

DCogent, 2020-02-24:
Thank you I will try this

DCogent, 2020-02-25:
I came across this error when i tried this. 

   if tag.DataType is 'Two_LetterSTR':
attribute error: 'str' object has no attribute 'DataType'

DCogent, 2020-02-25:
This is how im using it. 

`                 for tag in taglist:
                        if tag.DataType is 'Two_LetterSTR':
                            value = read_custom_string(tag,TagName)
                        else:
                            value = comm.Read(tag.TagName)
                        value = comm.Read(tag)                 
                        ValueList.append(value)`

evaldes2015, 2020-02-25:
What are you trying to do?
How did you generate taglist?

DCogent, 2020-02-25:
Im just grabbing it from a txt file

DCogent, 2020-02-25:
if i read in DataType: "Two_LetterSTR" i want to run it through def read_custom_string. The one dmroeder wrote above

evaldes2015, 2020-02-25:
If you're reading taglist from a text file, how is this a pylogix issue?

DCogent, 2020-02-25:
Well I found it wasn't really an issue.  Ended up being I just needed insight on how to read a custom string

DCogent, 2020-02-25:
Pylogix is working correctly. Is just sees the custom string as a UDT so it give me more data then I need.

dmroeder, 2020-02-25:
My function won't exactly work the way I described it, I made the assumption that your tag list came directly from GetTagList().  You would somehow have to know the difference between the base types and your custom string, then call my function when necessary.

DCogent, 2020-02-25:
I think im seeing that. When I say: tag = "the tag I know is using that string type" and then run it through your function it works. 
Is there anyway I can read the DataType prior to reading in the value?

DCogent, 2020-02-25:
So I got it working basically how I want. Thank you so much!! I am noticing though if I put a 1 character string through the function "read_custom_string"  I get a error: 

      value = ''.join([chr(d) for d in data])
TypeError: 'int' object is not iterable

All others work great. 
Just not sure how to fix it.

dmroeder, 2020-02-25:
Ah yes, silly me.  Maybe something like this:

```
def read_custom_string(tag_name):
    str_len = comm.Read('{}.LEN'.format(tag_name)).Value    
    data = comm.Read('{}.DATA[0]'.format(tag_name), int(str_len)).Value
    if str_len == 1:
        value = chr(data)
    elif str_len > 1:
        value = ''.join([chr(d) for d in data])
    else:
        value = ''

    return value
```

DCogent, 2020-02-25:
Thank you!!! 

I got that to work perfectly. 
Again Thank you so much for you time and effort. 

You can close this issue if you'd like.

dmroeder, 2020-02-26:
Glad it helped.  If you want some help, or advice in the future, feel free to email me.

closed 2020-02-26 by dmroeder

kodaman2, 2020-02-28:
Def need to add Reading Custom String to the docs. It seems like a recurring open issue.

chrisegge, 2022-10-01:
How about writing to a custom string that is un a UDT? STRING5 with only a length of 5, I am currently writing to the individual parts in SINT[5] and then setting the length with .LEN

sairamungarala77, 2024-12-05:
> Ah yes, silly me. Maybe something like this:
>
> ```
> def read_custom_string(tag_name):
>     str_len = comm.Read('{}.LEN'.format(tag_name)).Value
>     data = comm.Read('{}.DATA[0]'.format(tag_name), [...]

Hi,

I have a question regarding this. Why would a single character string read yield an integer but others (more that 1 character) yield an array of integers?

dmroeder, 2024-12-05:
Probably because of this:
https://github.com/dmroeder/pylogix/blob/master/pylogix/eip.py#L340-L341