dmroeder/pylogix #107: Trouble reading tags inside an array.  Possibly a UDT?

cstrutton, 2020-02-26:
I am a python programmer not a PLC programmer.  Your library is simply the greatest.  Thanks for all your hard work.  
I use RSLinx Classic to find out what tags I want to investigate.  

I was able to read DailyTotal:
[image]
using this code:
[image]

Unfortunately, this DailyTotal tag:
[image]
does not work with this code:
[image]
Am I addressing this tag wrong? Or am I missing something?

kodaman2, 2020-02-26:
Hey there, what happens if you do:

```
Program:Production.ProductionData.PrevDayHourlyParts[0].Total.DailyTotal
```

Next time please paste your python code rather than screenshots, and format with the code markdown:
https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code

dmroeder, 2020-02-26:
That is quite the UDT you have there.  It should be possible.  What does ret.Status print out?

Can you do this in the python interpreter
```
import pylogix
pylogix.__version__
```

cstrutton, 2020-02-26:
dmroeder:
Python version 3.8.1, Pylogix version 0.6.1
HourlyParts is a separate tree that I read in the first example.  
ret.Status is Path segment error
LOL...yes it is quite the UDT (if that is what it is)

kodaman2:
I would never have thought to try that but it works.  It even kinda makes sense.
I apologize for the code pics.  I had to take pics of RSLinx so I did the same with VSCode without thinking.

Thanks for everyone's quick responses.  This is a great project enabling guys like me to get even deeper in over their heads.  Love it!
That being said, I have access to a multitude of PLC (several hundred on my shop floor).  If I can help with testing, let me know, I would love to contribute if I am able.

dmroeder, 2020-02-26:
Oh duh, I see it now.  Got confused on which screenshot I was looking at.  Would you be willing to export that UDT and email it to me?  Assuming that there is actually no typo, I wonder if there is something with how long it is.

kodaman2, 2020-02-26:
> kodaman2:
> I would never have thought to try that but it works. It even kinda makes sense.
> I apologize for the code pics. I had to take pics of RSLinx so I did the same with VSCode without thinking. [...]

Yeah when expanding UDTs the array name gets repeated on the drilldown but for reading the tag you only need to know the index:

```
PrevDayHourlyParts[0]
```

dmroeder, 2020-02-26:
OOOOOHHHH, good call @kodaman2.   I missed that point in your original reply.  My game is off. 

Disregard sending me that @cstrutton, I didn't quite grasp that it worked when you tried what kodaman recommended.

dmroeder, 2020-02-26:
Actually @cstrutton , if you get time, send me an email, we can talk about what kind of hardware you have at your disposal.

dmroeder, 2020-03-10:
@cstrutton , if you want to help out you could check out issue #110 and give me feedback on routing.

closed 2020-03-11 by kodaman2