dmroeder/pylogix #109: Reading tags local to AOI

sdmolt, 2020-02-28:
In the PlantPAx AOIs, there is a ".Inf_Type" parameter (".HMI_Type" in newer versions) which has the datatype of the tag stored locally as a STRING_20. I want to use the returned datatype to determine what tags I want to read from the PlantPAx tag. However, it is not returning correctly. Here is what I get:

Code:
```
with PLC() as comm:
    comm.IPAddress = "192.168.1.2"

    ret = comm.Read('AnalogInputTest.HMI_Type')
    print(ret.Value)
```
Output: 
`b'\xb9\x8d\x05\x00\x00\x00P_AIn\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'`

The correct output should be:
`P_AIn`

This issue seems to only be with strings, I can read other tags which are numbers with no issue.

dmroeder, 2020-02-28:
See issue #106

sdmolt, 2020-02-28:
Completely missed that. Thanks @dmroeder

closed 2020-02-28 by sdmolt

dmroeder, 2020-02-28:
No worries!