dmroeder/pylogix #111: How to discover devices in all Vlans?

KaloyanR, 2020-03-10:
Is there a way to discover devices in  differents subnets?
or with a list of addresses not broadcaast msg?

dmroeder, 2020-03-10:
There was a related discussion going on in issue #93

KaloyanR, 2020-03-10:
ok, thanks
maybe broadcast msg is not good idea, but can I ask all devices one by one with a list of my IPs something  like for loop to present themselves:
IP, Product Name + Product Id + Vendor/Device ID ect.

dmroeder, 2020-03-10:
I recently added GetDeviceProperties() that will get the properties of a device a particular IP address:

```
from pylogix import PLC

with PLC('192.168.1.10') as comm:
    ret = comm.GetDeviceProperties().Value
    print(ret.ProductName)
```

dmroeder, 2020-03-10:
Or for a list of addresses, you might have to do something like this:

```
from pylogix import PLC

addresses = ['192.168.1.10',
             '192.168.1.11']

for address in addresses:
    with PLC(address) as comm:
        ret = comm.GetDeviceProperties().Value
        print(ret.ProductName)
```

KaloyanR, 2020-03-10:
OK, Thanks I will try it.

Traceback (most recent call last):
  File "C:\MotorService\testpylogix.py", line 14, in <module>
    getDevices()
  File "C:\MotorService\testpylogix.py", line 9, in getDevices
    with PLC(address) as comm:
TypeError: __init__() takes 1 positional argument but 2 were given

dmroeder, 2020-03-10:
I'm guessing address in your code is a list.  If you want help with your code, post it.

KaloyanR, 2020-03-10:
from pylogix import PLC

def getDevices():

    addresses = ['192.168.11.11',
                 '192.168.11.12']

    for address in addresses:
        with PLC(address) as comm:
            ret = comm.GetDeviceProperties().Value
            print(ret.ProductName)
                        
if __name__ == '__main__':
	getDevices()

KaloyanR, 2020-03-10:
OK It works. My fault in installation and venv settings.

dmroeder, 2020-03-10:
Okay, cool. Out of curiosity, what version of python and pylogix are you using?

```
import pylogix
pylogix.__version__
```

KaloyanR, 2020-03-10:
Pytnon 3.7
pylogix 0.6.2

But there is new problem, in given IP with no device ... error timeout!
How can i pass this to continue to next IP, cause i need kind of network inventory/discovery.

kodaman2, 2020-03-10:
@KaloyanR In regards to the timeout use a try catch.

https://docs.python.org/3/tutorial/errors.html

KaloyanR, 2020-03-11:
yes, 'except: pass' works, thanks.

kodaman2, 2020-03-11:
@KaloyanR np. Normally you'd want to handle the exception. Print to the console or log, or something else. I guess if the IP doesn't exist, there's nothing to do hence why pass might be acceptable in your case.