dmroeder/pylogix #113: How to get the list of a specific tag?

Ivan197, 2020-03-12:
Hello, how could I get the list of a specific tag, for example;
CTU_Segundos
       CTU_Segundos.PRE
       CTU_Segundos.ACC
       CTU_Segundos.DN
       etc.

I also have a question, what difference is there between the **tags.Value** from:
for t in tags.Value:
        print (t.TagName, t.DataType)

and the ret **(ret.Value)** of print():

ret = comm.Read ('CurrentScreen')
    print (ret.Value)

Will the **tags.Value** work as an index?

kodaman2, 2020-03-12:
Hi Ivan,

No, there's no function to return a timer tags. You have to know how to destructure the raw data.

Dustin has made an example 40_Read_Timer.py inside examples folder.

The read function returns a Response class which contains the following (.TagName, .Value, .Status)

Ivan197, 2020-03-12:
Hello kodaman2 ...
I thank you for the information, I will use the example 40_Read_Timer.py for the realization of my project.

Thanks and regards.

closed 2020-03-12 by kodaman2