dmroeder/pylogix #115: Running as a service

samiam9297, 2020-03-13:
My program runs great when I open it through Python IDLE / cmd line, but when I run it through AlwaysUP, I get the error:

`Traceback (most recent call last):
  File "C:\Monitoring\dataOps.py", line 766, in plcOps
    GPLCVals = dict(zip(mapPLCTags, mapPLCValues.Value))
AttributeError: 'Response' object has no attribute 'Value'`

Here is the method plcOps():
`# Get latest values from PLC
def plcOps():
    try:
        with PLC() as comm:
            comm.IPAddress = "192.168.1.12"
            mapPLCValues = comm.Read('Monitoring_Data[0]', 1000)
            global GPLCVals
            GPLCVals = dict(zip(mapPLCTags, mapPLCValues.Value))
    except:
        logOps.errorLogs.error(traceback.format_exc())
        logOps.infoLogs.error("plcOps failed")
        logOps.infoLogs.error("plcOps being recalled...")
        plcOps()

`
I'm using AlwaysUP to run it as a service.

dmroeder, 2020-03-13:
From AlwaysUP, see what version you are running:

```
import pylogix
pylogix.__version__
```

kodaman2, 2020-03-13:
@dmroeder lower case value suggests that is either the rogue repo or an old version installed before it was changed to upper case.

This is never going to end lol.

kodaman2, 2020-03-13:
On your alwaysup configuration check which python version you are using, then do pip list for that python version to check pylogix version, highly likely it will be 0.5.2. Or check your python idle version which runs the script fine, and use that version on your alwaysup configuration.

samiam9297, 2020-03-13:
@dmroeder @kodaman2 I've been playing with this for a bit.
In CMD:
`C:\Users\Admin>C:\Python376\python.exe
Python 3.7.6 (tags/v3.7.6:43364a7ae0, Dec 19 2019, 00:42:30) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylogix
>>> pylogix.__version__
'0.6.2'
>>>`

In AlwaysUp:
[image]

kodaman2, 2020-03-14:
From cmd prompt do:

```
pip list
```
What do you get?

samiam9297, 2020-03-14:
I was running 0.6.2. Its working now. Very confusing error, this was a freshly installed machine. Here is what I did:

1. cmd prompt > pip uninstall pylogix
2. follow install instructions on github homepage to reinstall
3. also reinstall without the --user tag, so:  'python setup.py install'

kodaman2, 2020-03-14:
Glad you got it working.

closed 2020-03-14 by kodaman2