dmroeder/pylogix #117: Multiple connections on same PLC (Micro850)

allanjonis, 2020-03-16:
Hi guys. I use pylogix for connect in on PLC and works well but now need more one script for connect on same PLC, when i do this, this message appear:

`...pylogix-master\pylogix\eip.py", line 1544, in recv_data
    part = self.Socket.recv(4096)
socket.timeout: timed out`

 Someone know how solve?
 Thank you!

dmroeder, 2020-03-21:
I would need to see your code but have you tried somerthing like:

```
from pylogix import PLC

plc1 = PLC('192.168.1.10')
plc2 = PLC('192.168.1.100')
```

allanjonis, 2020-03-23:
Thank you dmroader, with one script work well, my problem is when I use two scrips for read the same PLC, example:

file.py:
from pylogix import PLC

plc1 = PLC('192.168.1.10')

file2.py:
from pylogix import PLC

plc1 = PLC('192.168.1.10')

kodaman2, 2020-03-23:
@dmroeder Does the micro800 have a limit on open connections?

allanjonis, 2020-03-23:
> @dmroeder Does the micro800 have a limit on open connections?

On official documentation inform 16 max connections per device, I use less than 5.
[https://literature.rockwellautomation.com/idc/groups/literature/documents/um/2080-um002_-en-e.pdf](url)
Page 58.

ottowayi, 2020-03-23:
It may be an issue with having the same connection id.  pycomm creates a new random 4-byte id for every connection and I've seen no issues with having multiple connections to the same PLC.  Maybe at [this](https://github.com/dmroeder/pylogix/blob/0c8a3de5e3b32afe6297bc5dd5d87a16975ca12e/pylogix/eip.py#L997) line you should have it generate a random id.  It also does the same thing for the vendor serial number, but I'm not sure if that's required. But, it works for me so I'm not touching it.

dmroeder, 2020-03-23:
Interesting @ottowayi, as is, there is no issue making multiple connections to CompactLogix or ControlLogix.  My Micro820 is out on loan, I'm trying to get it back to test.  @allanjonis , if you want to try editing the line @ottowayi linked to generate a random value (like the line below), see if it makes a difference.

ottowayi, 2020-03-23:
Oh I guess I missed the part where it works for Compact/ControlLogix.  Then yeah that's probably not it.  I just did a quick compare between our forward open methods and that was the only real difference I saw.

dmroeder, 2020-03-23:
It is still very possible that it makes a difference for the Micro8xx series specifically.

kyle-github, 2020-03-24:
Please update this ticket if you can find out.   I do randomized connection IDs in my library, but the last time I had access to a Micro8xx was about five years ago.   I wish there was some sort of online "library" of PLC hardware.

I do know that using the same connection IDs, caused some problems with ControlLogix in some cases.  I think I was not shutting down the connection correctly, but reuse of the same ID was also a factor.

kyle-github, 2020-03-24:
One of the things I want to write (far too little time right now to make progress) is a simulator that I can build by feeding it actual packet dumps collected from people with real hardware.  I aim to have that become part of my test cycle.

dmroeder, 2020-03-24:
@allanjonis ,  I was able to borrow a Micro820 from my distributor, I have 2 files reading fine.   Can you do the following from the python interpreter:

```
import pylogix
pylogix.__version__
```

allanjonis, 2020-03-25:
@ottowayi ,I will try this today and inform you, thank you for the information.
@dmroeder ,here the version: 0.4.12.

dmroeder, 2020-03-25:
That is pretty old, grab the latest code and try it.  It's currently at 0.6.3.

dmroeder, 2020-03-25:
@kyle-github thought I'd share, I updated CCW to 12 and it has a Micro850 simulator.  Not sure when they added it but you can access it via Ethernet when it is running.

kodaman2, 2020-03-25:
@dmroeder Gonna re try this at some point. I had installed CCW a while back when I was doing the unittests and the sim would crash every time I ran a pylogix script.

dmroeder, 2020-03-25:
@kodaman2 I did the same 2 script test with the simulator and it worked.

kodaman2, 2020-03-25:
Okay, @dmroeder so as a work around everything works if we don't send ForwardClose?

dmroeder, 2020-03-25:
I only tested reading a DINT with one script and a STRING with the other, both worked fine running at the same time. The simulator appears to stay running as long as ForwardClose is not called, which means don't instantiate with a With statement or call Close().

That being said, I installed the version @allanjonis is running in a virtual environment and I can run 2 scripts at the same time connected to the simulator. 

@allanjonis , are you running this with the simulator or actual hardware?  I cannot recreate your issue, can you share your exact code?

allanjonis, 2020-03-26:
Hi, sorry for the delay, is one physical device here, now the version is updated, and the error appears again.
I started file1, changed the "CIPTOConnectionID"  like the @ottowayi said and start file2, no errors on last hour :)
 I will test for more some hours and back here, thank you for all.

dmroeder, 2020-03-26:
Excellent, thanks for the feedback. When it was not working for you, was it immediate or did it happen after a period of time?

allanjonis, 2020-03-26:
@dmroeder , stopped on the first second.

kyle-github, 2020-03-27:
@dmroeder thanks for the tip about CCW.  I'll have to figure out how to get that.  Weird about ForwardClose.   I know I was sending that back when I was testing against real hardware.  Perhaps I was causing reboots on those systems :-/

dmroeder, 2020-03-27:
@kyle-github I figured out what was causing the crash.  I was not waiting to receive the reply from my ForwardClose, immediately sending the Unregister session.  A bug in my code and a bug in the simulator if you ask me.  The actual hardware does not behave the same.

In CCW, if you go to the Tools drop down, you will see the Micro800 simulator.  Running it is surprisingly more straighforward than you'd expect.  Especially if you have used Emulate or SoftLogix :)

kyle-github, 2020-03-27:
@dmroeder, thanks for the update.  I do send a ForwardClose every time and I wait for a bit for a response.  However, I do not send a Unregister session.  I just close the connection.  I used to do that and probably should again.
This does sound like a bug in the simulator.

dmroeder, 2020-03-27:
I only send ForwardClose when a script exits, or the user can call it themselves.  I didn't listen for the reply, I just moved on to send the UnregisterSession.  I consider that a bug in my program, however I think if the simulator crashes because of this, then it it also a bug in the simulator.  It's possible that if I never sent that UnregisterSession, then the sim wouldn't crash, I'll test that just for fun, though I'll fix my program properly.

ottowayi, 2020-03-27:
I wasn't aware of the simulator, so I tested mine with it.  Mine worked with no issues, though it does wait for a response to the Forward Close before unregistering.  I would say it's more of a bug in the simulator, because I for a long time did not realize my forward close wasn't working and I never saw any issue with physical hardware. So now it get's the confirmation for the Forward Close as well.  I did find that the simulator will not work if your project file path has a [comma](https://rockwellautomation.custhelp.com/app/answers/answer_view/a_id/817678/loc/en_US#__highlight) in it for some reason.

allanjonis, 2020-03-27:
@dmroeder,  some times run for some minutes until the error happens.
 But the change in "CIPTOConnectionID" solved the problem.
 I will add a few lines in the script to change this field at each start.
 Thank you guys!

dmroeder, 2020-03-27:
Okay, I just didn't give it enough time. Thanks for the update, I'll get this fixed here soon.  And thank you @ottowayi, it probably would have been a while before I suspected that one...

kyle-github, 2020-03-28:
@dmroeder, thanks so much for the tip about CCW v12!  I finally got a chance to download it and try it out (I have never used it before) and it is great for testing!    My library does not seem to crash the simulator, so I guess I am handling ForwardClose correctly :-D   Now I can really dig in and see exactly what data types I get back for things like STRING.  IIRC it is different from what I get back from a ControlLogix.

kyle-github, 2020-03-28:
I ran across this on AB's Knowledge Base about limits on Micro800: [How Ethernet/IP explicit message instructions are handled in Micro800?](https://rockwellautomation.custhelp.com/app/answers/answer_view/a_id/566260/loc/en_US).   There is an underlying limit of 4, sort of.   It means that Micro800 PLCs are pretty limited.

kyle-github, 2020-03-28:
[This one](https://rockwellautomation.custhelp.com/app/answers/answer_view/a_id/566259/loc/en_US) has even more details.   Looks like you really want to limit the message flow to a Micro800.

dmroeder, 2020-03-29:
Interesting @kyle-github , my first instinct was that this was a connection limit issue.  I've done many tests with the Micro820 and I cannot re-create the issue.  Right now I have 5 scripts reading with no delay between reads, it's been going for an hour and a half.  I have no problems making the change to randomize that connection ID, it just bothers me that I cannot recreate it.  I get the feeling that there is something else going on.

kyle-github, 2020-03-29:
It is really unclear.  If you look at the first KB item, it looks like you can have up to 4 simultaneous connections all pushing requests at the Micro800.  Anything over that and you start to queue internally.   The second one says that you can process _up to_ ~10 messages per scan, so you really should not be too limited unless you are hammering the PLC.   But I cannot figure out how applicable the 4-connection limit is.  Is that connections to the PLC?  Just front-end queues?   AB, the gift that keeps on giving (migraines).

The lack of support for multiple request packets is a pain.

dmroeder, 2020-03-30:
@allanjonis ,

Something is not adding up here.  I ran 7 scripts connected to my Micro820 all night without issue.  All 7 scripts were reading with no delay, they ran for about 12 hours before I stopped them.

I know you said you updated to the latest version of pylogix, but can you tell me what python prints out as the version?  As of 0.6.3, it should not error in the same way as your original post if the connection times out, which makes me believe you are not actually running the latest version.  That may not matter either though, I ran 0.4.12 for about an hour with 4 scripts with no issue.

The _only_ way I could reproduce some kind of instability was to read with no delay and close the conection after every read, which is not the way to read data from a PLC. Could you share your code with me?  Either here or via email?

Although I pushed a commit to randomize the connection ID, I cannot demonstrate that the change actually helps anything.  I have no doubt that you are having an issue, I just can't confirm what it is.  If the connection ID is really an issue, I should be able to reproduce it.

allanjonis, 2020-03-30:
@dmroeder thank you for your efford.
 It can really be a problem due to my environment.
 Here I have 4 PLCs 850 and 8 scripts running at the same time, and some C # applications connected too.
But the change in randomizing the connection ID really solved the problem here.
  Here the code:

`from pylogix import PLC
import pypyodbc
import requests
from datetime import datetime
import datetime
import time

b1p2in1f = True
container11c = None
d = None
h1 = datetime.datetime.now()
n11 = 0

# PLC B1S
while True:
    with PLC() as comm11:#
#        try:
        now = datetime.datetime.now()
        d = (h1 - now).total_seconds()

        comm11.IPAddress = '192.168.10.51'
        p2in0 = comm11.Read('_IO_EM_DI_00')
        p2in1 = comm11.Read('_IO_EM_DI_01')
        p2in2 = comm11.Read('_IO_EM_DI_02')
        p2in3 = comm11.Read('_IO_EM_DI_03')
        p2in4 = comm11.Read('_IO_EM_DI_04')
        p2in5 = comm11.Read('_IO_EM_DI_05')
        p2in6 = comm11.Read('_IO_EM_DI_06')

        # B1S LaneID 11 Frontal
        if p2in1.Value == False and p2in4.Value == False:
            b1p2in1f = True

        if p2in6.Value == True:
            h1 = datetime.datetime.now()
            b1p2in1f = False

        if d <=12:
            b1p2in1f = True

        if  b1p2in1f == True and p2in1.Value == True and p2in4.Value == True and p2in6.Value == False:
            b1p2in1f = False
            n11 = n11 + 1
            print('Call F LANE11')
            #lane11F(lane11F)

        if n11 > 1000:
            n11 = 1

        # B1S LaneID 11 Container

        if p2in2.Value == False and p2in6.Value == False:
            container11c = True

        if p2in2.Value == True and p2in6.Value == True and container11c == True:
            container11c = False
            print('Call C LANE11')
            #lane11C(lane11C)

#        except:
 #           print('Error')
  #          pass

`

allanjonis, 2020-03-30:
It can really be a problem due to my environment.
Here I have 4 PLCs 850 and 8 scripts running at the same time, and some C # applications connected too.
But the change in randomizing the connection ID really solved the problem here.
  Here the code:

`from pylogix import PLC
import pypyodbc
import requests
from datetime import datetime
import datetime
import time

b1p2in1f = True
container11c = None
d = None
h1 = datetime.datetime.now()
n11 = 0

# PLC B1S
while True:
    with PLC() as comm11:#
#        try:
        now = datetime.datetime.now()
        d = (h1 - now).total_seconds()

        comm11.IPAddress = '192.168.10.51'
        p2in0 = comm11.Read('_IO_EM_DI_00')
        p2in1 = comm11.Read('_IO_EM_DI_01')
        p2in2 = comm11.Read('_IO_EM_DI_02')
        p2in3 = comm11.Read('_IO_EM_DI_03')
        p2in4 = comm11.Read('_IO_EM_DI_04')
        p2in5 = comm11.Read('_IO_EM_DI_05')
        p2in6 = comm11.Read('_IO_EM_DI_06')

        # B1S LaneID 11 Frontal
        if p2in1.Value == False and p2in4.Value == False:
            b1p2in1f = True

        if p2in6.Value == True:
            h1 = datetime.datetime.now()
            b1p2in1f = False

        if d <=12:
            b1p2in1f = True

        if  b1p2in1f == True and p2in1.Value == True and p2in4.Value == True and p2in6.Value == False:
            b1p2in1f = False
            n11 = n11 + 1
            print('Call F LANE11')
            #lane11F(lane11F)

        if n11 > 1000:
            n11 = 1

        # B1S LaneID 11 Container

        if p2in2.Value == False and p2in6.Value == False:
            container11c = True

        if p2in2.Value == True and p2in6.Value == True and container11c == True:
            container11c = False
            print('Call C LANE11')
            #lane11C(lane11C)

#        except:
 #           print('Error')
  #          pass`

allanjonis, 2020-03-30:
It can really be a problem due to my environment.
Here I have 4 PLCs 850 and 8 scripts running (two per device) at the same time, and some C # applications connected too.
But the change in randomizing the connection ID really solved the problem here.
 The Code, and one print screen with one error if I no change the ID:
[https://transferxl.com/06vhjkfXrnL6jz](url)

dmroeder, 2020-03-30:
Well that's great then.  If you get time, I made the change to 0.6.5, could you try it out and let me know?  Thanks for haning in there!

allanjonis, 2020-04-07:
@dmroeder sorry for the delay (hard times).
The version 0.6.5 is working very well here, thank you!
 I can close the issue guys?

dmroeder, 2020-04-07:
@allanjonis , no worries buddy, thanks for the update!

closed 2020-04-07 by dmroeder