dmroeder/pylogix #118: Scan VPN network for PLCs

AndrewStuhr, 2020-03-16:
Is Discover() designed to scan for PLCs on VPNs?  If not, do you have any plans to?  Or do you have any recommendations for how I could modify pylogix to do so?

Thanks

kodaman2, 2020-03-17:
Can you give a bit more context of what you are trying to do, surely the PLCs are on a LAN, right?

Are you saying you connect to your plant thought VPN and you want to run pylogix remotely? If you are, I don't see how that use case can be beneficial to pylogix, plus it would probably be a security issue to say the least.

What I recommend is that you setup a script that can securely connect to a computer onsite that is on LAN, to trigger Discover, and then send you the results back somehow, but pylogix through VPN is most likely out of scope for this project.

AndrewStuhr, 2020-03-17:
I'm trying to use pylogix to display PLC values on a webpage.  Ideally, I could display data in PLC on my work's network while connected over VPN, or display a customer's PLC values while VPN into their network.

I understand that it's a particular request with security issues, and probably outside pylogix's scope.  Just wanted to check though, thanks!

kodaman2, 2020-03-17:
Thanks for the details. You need to write some backend code api, if you want to use python still I recommend flask or Django. Then request data from your local api on your web application.

You could use other stacks like node.js and react, etc. Depending on how much data you are displaying you could write a python script with pylogix, that outputs data to a json file, convert script to exe with pyinstaller. Then from node.js call that script with whatever interval you want. Setup an api so that you can request that json file over the web. Obviously if is a lot of data you'll need to think this through. 

https://programminghistorian.org/en/lessons/creating-apis-with-python-and-flask

No matter what you do, your problem is not pylogix related. You need a backend api no matter what.

kodaman2, 2020-03-18:
Here's a super simple example of flask with pylogix, I made two routes. `app.py` will need to be ran on your LAN, and then you can access remotely by knowing your external ip or dns address you'll have to open whatever port want flask does 5000 by default. You'll still need to figure out user authentication and such. I may or may not expand in the future, depends if more people would want to use something like this.

https://github.com/kodaman2/pylogix-api

closed 2020-10-01 by dmroeder