dmroeder/pylogix #119: C++ wrapper

Folcogh, 2020-03-26:
Hi, and first thanks for this library.
Would it be possible to have a C++ wrapper for it ? It's my main language by far, and I would want to avoid learning Python just for that...

Thanks in advance for your answer ;)

dmroeder, 2020-03-26:
Well I don't want to learn C++ to create a wrapper so that you don't have to learn python :)  Please understand that I say that with a bit of humor.  

I have no plans to write wrappers for any other languages.

kodaman2, 2020-03-26:
Isnt't @kyle-github  https://github.com/kyle-github/libplctag what you need?

kodaman2, 2020-03-26:
> Well I don't want to learn C++ to create a wrapper so that you don't have to learn python :) Please understand that I say that with a bit of humor.
> 
> I have no plans to write wrappers for any other languages.

Haha well, someone has to learn one of the other, and it ain't going to be me lmao. In all seriousness, libplctag should be what he needs.

dmroeder, 2020-03-26:
Yeah good call on Kyle's project, check that one out @Folcogh .

Folcogh, 2020-03-26:
Don't worry dmroeder, I understand.
Thanks all for your hint about libplctag, but reading the available functions, I prefer the pylogix API.
I will learn Python if I really need it.

Thanks again for your attention =)

dmroeder, 2020-03-26:
@Folcogh , you should definately learn python though, it's awesome :)

If you end up giving it a shot and want any pointers, send me an email, I'll do what I can to help.

Folcogh, 2020-03-26:
Oh, really nice, thank you very much !

kodaman2, 2020-03-26:
Here's a quick intro to pylogix, I made a while back.  https://www.youtube.com/watch?v=GssCEQf1qxM and remember the sample folder is pretty good too.

Python is a breeze if you know C++. Just whatever you do use a decent IDE because indentation will be the most difficult thing for you in the beginning. VSCode, Intellij IDEA with python plugin come to mind.

kyle-github, 2020-03-27:
Ooh, nice, @kodaman2!

@Folcogh what kind of API do you prefer?  I am always looking to expand the use of my library too :-)   For some reason, most of the users of my library seem to be C# programmers.  At least the ones that I hear from.

kodaman2, 2020-03-27:
@kyle-github Thanks. C# users probably because is easier to make gui applications with dot.net, WPF and good ole' WinForms.

Folcogh, 2020-03-27:
@kodaman2 thank you for your work, I'll take a look!

@kyle-github I want to read/write tag of a running program on various machines, so a must-have feature is to retrieve the list of existing tags. Without that, a library is useless for my needs.
Another useful thing would be to read a tag using its name, instead of a returned internal handle, else I don't see any way to use tags of an external PLC program.

I have seen your API ( https://github.com/kyle-github/libplctag/wiki/API ), but I don't see any function like that.

Nevertheless, thanks for your job, and if you want, we can continue this discussion on your library board. :)

Have a nice day all!

closed 2020-05-05 by kodaman2