dmroeder/pylogix #123: Support for CIP Read-Modify-Write command (0x4E) with Micro8xx?

kyle-github, 2020-04-06:
Question for @dmroeder, I am implementing the RMW command to support individual bit selection from e.g. DINT tags.   I got it all working nicely against my Control Logix PLCs.   Then I tried the Micro8xx emulator that you (thanks!) mentioned.   Major Whiskey Tango Foxtrot moment: it appears that (the emulator at least) the Micro8xx treats the mask parts of the command as big-endian!?!

Have you run into this?

kyle-github, 2020-04-06:
I got the emulator to work by reversing the byte order of the masks.   But... really??

dmroeder, 2020-04-06:
Huh, I would have guessed that the Micro800 wouldn't have supported it.  But reversing the byte order, I'm not surprised, because Rockwell.  Let me give you an example.  Until recently, none of the Rockwell controllers supported unsigned integers or 64 bit math, until the Mirco800 series.  What?  Then comes along the high end 5380 and 5580 controllers, only v32.  So the lowest end and highest end (with the latest firmware) support what should be standard features.

dmroeder, 2020-04-06:
BTW, thanks, I'll test this out when I get a moment, I have a Micro820 on loan.

dmroeder, 2020-04-06:
@kyle-github this behavior seems to be only with the simulator.  Actual controllers don't behave this way.  I'm going to submit a couple of bugs to Rockwell about this and how ForwardClose is being handled.

kyle-github, 2020-04-07:
Thanks very much for testing this out!  I'll revert my code changes.   I generally try not to swear in code comments but...

kyle-github, 2020-04-07:
Thanks again.  I really appreciate the feedback on this weird problem.  Good to know that it is the emulator not the hardware.

closed 2020-04-07 by kyle-github