dmroeder/pylogix #124: packet size negotiation

kyle-github, 2020-04-07:
I wanted to share how I negotiate packet sizes.  Some PLCs support larger packets and some do not.   So far, the unifying theme has been that if the CPU is newer (an L55 does not) _and_ the network module firmware and CPU firmware are sufficiently new (>v20 on the CPU firmware, varies on the network module), then you can negotiate a larger packet.  In some cases you can actually get an error that tells you what the PLC will accept.

It is a bit annoying, so there is probably a cleaner algorithm.

I start by sending a Forward Open Extended (or Expanded, not sure what the right name is).   This is just like the normal one, but the two fields that negotiate the request and response sizes are 32-bit instead of 16-bit.  The meaning of all the fields is the same but the size field is now 16 bits instead of 9 bits.

If I get a CIP error code 0x08 (Unsupported), then I drop back and retry using an old Forward Open on the second try.  However, if I get a CIP error 0x01 then I dig a little deeper.   If you get 0x01 as the main CIP error byte and there are two additional words (not bytes!) of status, then I read the next two words.  The first is the additional status.

If that additional status word is 0x109, then the next word appears to be the size that the PLC will accept for a request packet.   It matched 4002 exactly which seemed to be a bit of a coincidence.

So the bytes are:

0xDB - Forward open extended response
0x00 - padding of some sort.  Seems to be zero.
0x01 - error code 0x01
0x02 - two additional words of error status.
0x09
0x01 - error code 0x109
0xA2
0x0F - 4002 in hex

I have tried this against a ControlLogix using an ENBT network module.  The CL was running firmware v15 and Forward Open Ex did not work and I got 0x08.   I tried against a CL that was running v20 and it did not work.  I tried  another with v21 or 24 (I can't remember) and and EN2T which had the last/latest firmware at that time (circa v24 of the CPU firmware) and it did work to negotiate a 4002 packet.  I tried with larger packet sizes and that is when I discovered the above error behavior.  The PLC would respond with the error above but tell me that it wanted a 4002-byte payload.  I tried with smaller numbers and it worked as well IIRC.  

 I have a chassis with a L55, L61 and L81, and a couple of ENBTs and it works with the following combinations:

L81/L81 network port - works fine, negotiates 4002 bytes.
L81/ENBT - Does not work.  I get 0x08 and negotiate to 500 or whatever it is bytes.
L61/L81 network port - works fine, negotiates 4002 bytes.
L61/ENBT - does not work, negotiates 500 bytes.
L55/* - never works with Forward Open Ex and only supports 500 bytes.

I only try Forward Open Ex with PLCs that I think are Control/Compact Logix.  I do not attempt them with PLC/5 or SLC or MicroLogix.

ottowayi, 2020-04-07:
I do something similar.  I didn't really find any other sizes besides 500/4000, so instead of a connection size, I have a `large_packets` argument.  But, it's no longer required since it by defaults tries and extended forward open, if that fails it tries the standard one.  If the standard fails, it just raises an exception.  Actually, I just use a [decorator](https://github.com/ottowayi/pycomm3/blob/ae677ca167e2fbec2840c3bef16aace0f7a1393a/pycomm3/clx.py#L52) to ensure a forward open is performed for whatever methods require one.

kyle-github, 2020-04-07:
Yeah, I try to short circuit it by trying 4002 bytes with FOEx first.   If it is a relatively new PLC, it just works.

dmroeder, 2020-04-12:
Thanks for the info gentlemen!  I like the idea of assuming large connection, fall back on small connection, give up.

Currently, my project will let you put in any value for connection size between 500 and 4000.  I spent a bit of time making sure that if someone entered a connection of 2000, for example, and the PLC actually supported large connections, that it would work.  It currently does.  The only reason that I handled the sizes between 500 and 40000 is because I figured if you could, they will try....

kyle-github, 2020-04-12:
I have yet to find any Logix-class PLC that supports any sizes other than 4002 or 508.   That is not to say that they do not exist, but so far, those are the only two I have seen.  

I do not allow users to negotiate packet sizes currently.   Since I am slowly gearing up to support small embedded systems (think 32kB of RAM), I will need to do that eventually.   Right now the library hides all the negotiation and just tries to grab the largest packet size it can.

dmroeder, 2020-04-12:
Any PLC that supports "Large Forward Open" will support connection sizes between 508 and 4002 bytes.  You can specify the connection size and it will accept it.  You can try it with pylogix if you want.

No why anyone would want to specify a connection size of 2000, for example, who knows.  It probably would have been better just to make it a boolean selection, large or not large.

kyle-github, 2020-04-12:
I was typing too fast.   I have yet to find any PLCs that do not support either a maximum of 4002 bytes or 508 bytes.    You can negotiate lower, but why?

I generally see between 5-10ms for response times from a higher-end PLC for a small packet.   At that rate, the processing overhead of the packet dominates the transmission time.   So larger packets are better.

closed 2020-05-12 by dmroeder