dmroeder/pylogix #125: Scanning Members of UDT

AndrewStuhr, 2020-04-20:
I was reading over issue #45 : "reading a tag of type UDT".  When dmroeder mentioned that there's never really been an intention to read the entire contents of a UDT, was that referring to reading the values in a UDT or the member names in the UDT (or both)?

I'm hoping to read the names (but not values) of a UDT's immediate children.  Is there any way to do this without an l5x file?  If not, do you know what it would take to modify pylogix to do this?

ottowayi, 2020-04-20:
I'm not sure that status of reading nested structures in pylogix, but I have what you're looking for in [pycomm3](https://github.com/ottowayi/pycomm3).  It will give you both the fully nested structure of a UDT as well as allow reading of the entire UDT too (assuming no External Access = None attributes).  If that functionality doesn't exist, you may be able to reference my code to get it working in pylogix as well.

AndrewStuhr, 2020-04-20:
Sweet!  I was able to read UDT contents with pycomm3.  I'm still curious what the status is with PyLogix and/or what it would take to add this functionality.

I'm going to start a separate thread in the pycomm3 issue's section.