dmroeder/pylogix #127: CIP Connection Class

raurban, 2020-04-21:
Hello. It looks like pylogix establishes a CIP Class 3 "explicit" connection with the ControlLogix PLC. Is it possible to establish a Class 1 "implicit I/O" connection with the ControlLogix using pylogix?

Thank you.

dmroeder, 2020-04-21:
Unfortunately you cannot establish class 1 connections., Class 3 only.

kyle-github, 2020-04-22:
Note that implicit data, such as used by produced/consumed tags, uses a form of Forward Open that I have not fully decoded, but then it usually requires multicast UDP support.   In Python, that's probably not too horrible.  

In order to keep the connection open, you need to keep a heartbeat from the client to the server.   What I normally see is that these need to be sent every 500ms, but I think that is part of the negotiation in the Forward Open.

I suspect, but am not sure, that implicit I/O works the same way.   Of course, this is Rockwell, so it is probably 95% similar except that every single thing is slightly different.

raurban, 2020-04-22:
Thanks for the replies.

Produce/Consume would be an excellent feature for pylogix.

When I'm working on a large automation upgrade with lots of drives, I typically offload drive simulations into a temporary ControlLogix processor and use produce/consume to the main PLC. The main PLC simply thinks its talking to a drive with real feedback. It would be awesome to do the same but with pylogix running on a SBC.

Also, for high speed synchronous data logging, produce/consume is the way to go.

Looks like Rockwell has given their blessing to OpENer Ethernet/IP open source stack project, which I'm sure you guys are familiar with. I'm curious if this is of any help to the pylogix project.