dmroeder/pylogix #128: Blank Type in GetTagList

AndrewStuhr, 2020-04-25:
I'm grabbing tags with comm.GetTagList().  I noticed a few tags have a blank type ("").  What does this mean?  Can a PLC tag not have a blank type?

dmroeder, 2020-04-25:
A PLC cannot have a blank tag.  Are you using the very latest (0.6.7)?

AndrewStuhr, 2020-04-25:
I'm using 0.6.6, but I'm guessing this is still odd behavior.  I'll upload my code and response.

dmroeder, 2020-04-25:
That is fine, I just wanted to make sure that I didn't goof anything up with the latest commits.

Is it the tag name that is blank or the data type?

AndrewStuhr, 2020-04-25:
Data type

dmroeder, 2020-04-25:
Well, when requesting the tag list, the controller will send things back that you wouldn't consider a tag.  Program names for example.  There are other items that I would consider noise that I try to filter out.

What is "MDXF1" in your program?

AndrewStuhr, 2020-04-25:
I'm actually not sure, it's somebody else's program.  I can probably find out on Monday though.  I was able to connect to the PLC us Kepware, and I could see values in this structure using their OPC quick client, so apparently this thing has data.

dmroeder, 2020-04-25:
So I guess the answer is yes, there can be things are sent by the controller that do not have a data type.  It would be interesting to know what MDXF1.

AndrewStuhr, 2020-04-25:
Sounds good, I'll send you the details Monday.  Thanks for the quick response!

dmroeder, 2020-04-25:
Sure thing.  So other things that come back with that CIP object are programs names, routine names, task names, etc.  Those wouldn't have data types associated with them.

AndrewStuhr, 2020-04-29:
I believe the type of this tag was an add-on-instruction or add-on-definition.  It started with add-on.  Does this make sense?

dmroeder, 2020-04-29:
It's very possible.  An AOI definition could come back when requesting the tag list, it would not have a data type.  The actual instance of the AOI would have a data type, which would be the AOI definition.  I think the name of the AOI definition would show up like "UDI:MyAOIName"

dmroeder, 2020-10-01:
The latest will filter out AOI definitions.

closed 2020-10-01 by dmroeder