dmroeder/pylogix #132: Endianness issue in Jython

JordanCClark, 2020-05-13:
## Versions
- pylogix: 0.6.8
- python: Jython 2.7 (Ignition)
- OS: Win10

Since Jython uses Java to emulate python libraries, the endianness of native pack/unpack is reversed.

Setting self.CIPTypes tuples to force little endian fixes the issue, and still seems to work under Python 3.8

```
        self.CIPTypes = {0: (0, "UNKNOWN", '?'),
                         160: (88, "STRUCT", '<B'),
                         193: (1, "BOOL", '?'),
                         194: (1, "SINT", '<b'),
                         195: (2, "INT", '<h'),
                         196: (4, "DINT", '<i'),
                         197: (8, "LINT", '<q'),
                         198: (1, "USINT", '<B'),
                         199: (2, "UINT", '<H'),
                         200: (4, "UDINT", '<I'),
                         201: (8, "LWORD", '<Q'),
                         202: (4, "REAL", '<f'),
                         203: (8, "LREAL", '<d'),
                         211: (4, "DWORD", '<I'),
                         218: (1, "STRING", '<B')}
```

kodaman2, 2020-05-13:
@JordanCClark Thanks for opening the issue, we'll investigate. 

- Is this issue experienced only with latest version? 
- If you have been using ignition for a while did you experienced this issue before? 

@dmroeder I thought we had a good share of ignition users, would seem odd that no one would report it until now.

JordanCClark, 2020-05-13:
I helped do some testing on the early versions. Thought I'd take a look again, now that I have a bit of extra time on my hands.

And to be fair, I'd say most Ignition users are reading/writing via OPC functions. Ignition just happens to be my most convenient Jython installation. Even then I had to adjust the imports since Ignition doesn't always handle relative imports very well. 

Let me spin up a VM with Jython by itself. I'll let you know what I find.

JordanCClark, 2020-05-13:
Confirmed with 
* Jython 2.7 standalone under Java 8/Ubuntu 18.04
* Jython 2.7 implelemted in Igntion v8/Java 11/ Win10 and Linux 18.04. 

Java uses big-endian across all platforms, so that it doesn't care (assuming Java-only applications) what the native endianness is.

[https://howtodoinjava.com/java/basics/little-endian-and-big-endian-in-java/](url)

kodaman2, 2020-05-13:
Thanks for the details. At first glance it seems like the issue is jython related.

@dmroeder We haven't explicitly said we support Jython. How do you feel about this patch, assuming it doesn't break any current python unit testing?

dmroeder, 2020-05-13:
@kodaman2 I did some quick testing and it doesn't seem to have any negative impact so if it helps the great @JordanCClark, I'm on board.

kodaman2, 2020-05-13:
I'll push patch right after I push small change to `PylogixTests.py` it broke after structuring the tag and response classes to their own file.

dmroeder, 2020-05-13:
Shoot, that was my bad (obviously)

kodaman2, 2020-05-13:
> Shoot, that was my bad (obviously)

No worries, you sent me the new structure too, and I missed it.

@JordanCClark  please install from head until there is a release available. Let us know how it went.

```
pip install git+https://github.com/dmroeder/pylogix
```

JordanCClark, 2020-05-14:
Looks good, gentlemen. Many thanks for the support!

closed 2020-05-14 by kodaman2