dmroeder/pylogix #135: Moxa Switch Response

AndrewStuhr, 2020-05-20:
A MOXA switch (SDS-3008) is responding to Discover() with these properties:

[image]

However, I get a timeout with null data when I run:
```
	with PLC() as comm:
		comm.IPAddress = '192.168.127.253'
		prop = comm.GetModuleProperties(1)
		print(prop.Value)
```

Do you know how I can query this device;s properties using GetModuleProperties() instead of Discover()

AndrewStuhr, 2020-05-20:
Also, the following device is not responding to GetModuleProperties(slot), yet it is returned by Discover() along with properties : '1763-L16BBB B/9.00'

kodaman2, 2020-05-20:
Regarding your first question, please read the NB notes in the readme. In short GetModuleProperties is for modules supported by Controllogix / Compactlogix lines. Readme also explains why Discover gets some information. 

Regards to the second question pylogix doesn't support Micrologix as far as I know. Also on the readme.

dmroeder, 2020-05-20:
As @kodaman2 says, GetModuleProperties(slot) is for devices that are in a chassis, like ControlLogix, CompactLogix modules and Point I/O modules.  For quering individual devices, like PowerFlex drives, use GetDeviceProperties().

AndrewStuhr, 2020-05-20:
Gotcha, thank you both!

AndrewStuhr, 2020-05-20:
Is there a way to determine if a PLC is a micrologix from these Discover() properties (so I can filter it out):

[image]

dmroeder, 2020-05-20:
You could probably do something like:

```
ret = comm.GetDeviceProperties().Value
if ret.ProductName.startswith("1763"):
    # Micrologix 1100
```

AndrewStuhr, 2020-05-20:
Thanks - do you know if this an exhaustive list of micrologix families that are not compatible with pylogix: '1763', '1762', '1766'

dmroeder, 2020-05-20:
Pylogix isn't compatible with any micrologix.  It's not the hardware that is the problem, it's the protocol.  For pylogix, there isn't really any difference at the protocol level between how to connect to ControlLogix or CompactLogix.  Micrologix architecture is different.  If it has data files, pylogix cannot read them (N:7, B:3, etc).

Some of the newer micrologix controller implement enough of the Ethernet I/P protocol to be discovered automatically by RSLinx, nothing more.  That is why pylogix can discover it, but you cannot read/write.

AndrewStuhr, 2020-05-20:
I just meant more of a general micrologix question - like do you know the other family numbers.  In your example you used 1763 since that was the micrologix I posted, but I was wondering if you know the other family numbers.  1762 and 1766 are the other family numbers I found online:

https://ab.rockwellautomation.com/Programmable-Controllers/MicroLogix-Systems

I'm trying to get a list of all micrologix family numbrers so I can filter them all out of the Discover() results.

dmroeder, 2020-05-20:
I don't use them but I think only the micrologix 1100 and 1400 support basic Ethernet I/P. So 1763 and 1766. I don't think any others have Ethernet.

closed 2020-10-01 by dmroeder