dmroeder/pylogix #136: Read/Write multiple tags when tags are array

ramezanifar, 2020-06-02:
### Preflight checks

- [x] Have you tried the examples?
- [x] Have you tried [pylogix-tester](https://github.com/kodaman2/pylogix-tester)?
- [ ] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [x] Question
- [ ] Other

## Description of issue

Hello;
I was trying to read/write multiple tags while tags are arrays.
Example:
```
tag_list = ['array1', 'array2']
```
where array1 is SINT[100] and array2 is SINT[100]

Now if I read the tag list in one attempt, it returns only the firs index of each array:
```python
ret = comm.Read(tag_list)
for r in ret:
   print(r.Value)
```
Is there a way to pass the length?
I tried to sum up the length of all tags and use something like:
```python
ret = comm.Read(tag_list, 200)
```
But still the same problem.

In addition to the read, I could not figure out multiple write when tags are array.
Can you please give me a hint?
Thank you in advance

dmroeder, 2020-07-17:
First, apologies for the long delay.

Short answer, you cannot read arrays with lists of tags.

Long answer:  at some point I added the ability to provide the data type with with a single tag read or a list of tags to read.  This helped improve performance for people that read a lot of unique tags because it eliminates getting the data type first.  So when you attempt to do something like this:

```python
tags = [('tag1[0]', 10), ('tag2[0]', 10)]
ret = comm.Read(tags)
```

I'm assuming that the 10 in those cases is the data type.  It would then skip getting the data type, assuming that it was "10", and get the values for tag1[0] and tag2[0]. That is why you only get the first element.

I made some changes the other day to take advantage of the multi-service message object in order to bundle getting the data multiple data types into one packet.  This speeds things up quite a bit for reading unique tags.

This doesn't help you though when trying to read lists of arrays.  For now, only read non-arrays in lists, read arrays individually to get the best performance.  I have functioning code to read arrays with lists, but I will have to figure out 2 things:  Using multi-service messages (mulit-read) with arrays larger an a single packet, and what to do about people that are currently providing the data type to bypass discovering the data type.  In other words, I'll probably have to make a change that would break compatibility in to accommodate this, which I don't like to do.

sairamungarala77, 2025-10-02:
I have a similar issue like this. When i try to do the following

`
REAL = 202
DINT = 196
BOOL = 193
INT = 195
STRING = 218
tags = [
    [middleware + ".TxTriggerID", INT],
    [middleware + ".TxMessage.Code", STRING],
    [middleware + ".TxMessage.LaneAssigned", INT],
    [middleware + ".TxMessage.LaneActual", INT],
    [middleware + ".TxMessage.Reason", INT],
    [middleware + ".HostLookupTime", DINT],
]
comm.Read(tags)
`

I get the following error 
`error('unpack_from requires a buffer of at least 4 bytes for unpacking 2 bytes at offset 2 (actual buffer size is 2)')`

I was able to Read a single tag with no issue but doing a Read on the list of tag arrays, i get the above error. The weird thing is that I was able to read the array of tags for a long time now. 

Any suggestions on this?