dmroeder/pylogix #137: Pylogix/Kivy/Buildozer APK issue

pmpn8ez, 2020-06-04:
## Type of issue
- [X ] Question

## Description of issue
Trying to get PyLogix working in an android app using Kivy / Buildozer
Seems to crash the second i try to read. Works FANTASTIC (great job dude!!) in windows/ as a compiled exe and as python, also in the android app Pydroid 3 as a python file. The second i compile it into an APK it seems to break it.

## Expected behavior
Should read the PLC as it does in windows
## Actual behavior
Crashes app on first read
## Code
Just simply trying:
from pylogix import PLC

clx = PLC('192.168.2.190',2)
read = clx.Read('tag')

works in every way until i pack it into an APK using buildozer

buildozer spec has pylogix as a requirments
Any suggestion on what it needs to be happy?

## Versions

Include versions to

- pylogix: newest (0.6.10)
- python: newest ( whatever buildozer download for python 3)
- OS: Android 10

kodaman2, 2020-06-04:
Hey there, might wanna check out the crash logs for your android app or setup a debug session to live monitor it.

If I was to guess, apk packaging is not importing pylogix properly. Or if is packaging properly, is crashing due to unable to connect to the PLC, could be permissions on the manifest too.

Best thing to do here, is to make sure this code has a try/catch and that you log everything so that you can read the logs. Or just hook up phone to laptop and setup an adb debug session that way.

```python
try:
    from pylogix import PLC
except Exception as e:
    # log here, exit program if pylogix module not found

try:
    clx = PLC('192.168.2.190',2)
    read = clx.Read('tag')
except Exception as e:
    # log here
```

With all that being said, this is not a pylogix issue.

pmpn8ez, 2020-06-04:
Pretty sure the import works as the app doesn't crash right away, Happens on the read.
I tried looking at the logs through buildozer/adb and they were so huge i couldn't spot were the app crashed, have to try again. In the 'build' folder for the app I do see the packager is downloading pylogix anyway.

For permissions, is there a wifi access permission to send/recieve data? that cold be a cause...

I'll try the try/catch and see what i can come up with.

I am very aware its not a pylogix thing, just wondering if anyone has any ideas or have tried making an APK with it. Like I said, i am a PLC guy and love tinkering with things like this. The simplest way I have ever seen to get info from a plc to a PC. I have made pretty extensive things in VB with OPC and DCOM, excel with DDE and with AdvancedHMI and that took a lot more work.

kodaman2, 2020-06-04:
> Pretty sure the import works as the app doesn't crash right away, Happens on the read.
> I tried looking at the logs through buildozer/adb and they were so huge i couldn't spot were the app crashed, [...]

Not judging your knowledge or experience, I just made that note, because you might get better answers with kivy/bulldozer community. Most people here are plc guys including myself with a little bit of android experience.

Nice, so import is good. Then crash is most likely a permission assuming both phone and plc are on the same subnet.

Have a look here https://developer.android.com/training/basics/network-ops/connecting I don't know if there is a special permission for wifi, but with the try/catch hopefully it logs it. Written a few apps with Java/Kotlin you could use the logging library to see issues like this, you would see permission issues easy. I don't know if you can do it with kivy/buildozer though.

pmpn8ez, 2020-06-04:
kodaman2! You are a Lifesaver!!! That was it. just had to add INTERNET to the buildozer.spec file to add the access to the network. Now it works in linux/windows/python and android.
Tried with my small project then was able to get a much larger program functioning with the same change.

closed 2020-06-04 by pmpn8ez

kodaman2, 2020-06-04:
Nice! Love it when is an easy fix!