dmroeder/pylogix #138: Question for device 5069-L310ER COMPACTLOGIX

kajatesar, 2020-06-08:
### Preflight checks

- [ ] Have you tried the examples?
- [ ] Have you tried [pylogix-tester](https://github.com/kodaman2/pylogix-tester)?
- [ ] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [ ] Question
- [ ] Other

> Delete items that do not apply below.

## Question

Hello, is your library compatible with the following device?

**PLC: 5069-L310ER COMPACTLOGIX 1MB**

Thanks very much!

kodaman2, 2020-06-08:
Hello, yes, it is.

closed 2020-06-08 by kodaman2

kodaman2, 2020-06-08:
By the way is probably quicker to  use pylogix tester to get this answer too, but as a rule of thumb all compactlogix and controllogix are supported through etherip protocol.

kajatesar, 2020-06-09:
@kodaman2 Yeah, I know but unfortunately I don't have the device yet. Thank you!

kodaman2, 2020-06-09:
> @kodaman2 Yeah, I know but unfortunately I don't have the device yet. Thank you!

No problem, when you get your plc, if you have any problems let us know.