dmroeder/pylogix #142: Reading each array of a UDT - versus - reading the UDT containing arrays

OwenDavid, 2020-09-05:
### Preflight checks

- [ y] Have you tried the examples?
- [ ] Have you tried [pylogix-tester](https://github.com/kodaman2/pylogix-tester)?
- [ y] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [y ] Question
- [ ] Other

> Delete items that do not apply below.

## Description of issue
Hi,

Following comments and advices from other issues that I have read here I have reduced the number of individual pylogix reads by moving my tags into three arrays inside the PLC memory. These are in one UDT item.

So to transfer 280 tags pylogix just has to request three arrays!  The performance increase has been fantastic and makes using the library a viable solution - so thanks!
 
Using three separate reads of the UDT components and specifying array length works fine, i.e.

ret = comm.Read(REAL[200], 200)
ret = comm.Read(BOOL[224], 224)
ret = comm.Read(DINT[7], 7)

Due to my PLC scan time of (11ms)  and network latency the three reads take nearly 30ms.

When I try to read the UDT as one object, i.e.

ret = comm.Read(MyUDT)

The performance is enhanced again. My roundtrip time is around 8-10ms. 

However.. I struggle to interpret the results of ret.Value after reading the UDT and have several queries.

The ret.Value appears to contain several copies of the first (REAL[200]) array. For example,

MyUDT comprises: REAL[200], BOOL[224], DINT[7].

The ret.Value of comm.Read(MyUDT) returns nine data fields. It seems several of the fields contain the same data with different offsets.

Here is the bytecode for the 9 data fields ( the arrays are largely empty in this example.)

ret = comm.Read(MyUDT)

print len(ret.Value)
>>> 9 

print ret
>>>MyUDT
['U=X9\xb4=\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00X9\xb4=\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\xe4\x07\x00\x00\t\x00\x00\x00\x05\x00\x00\x00\x11\x00\x00\x00%\x00\x00\x00\x05\x00\x00\x00\xf9\xab\x02\x00', '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00X9\xb4=\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\xe4\x07\x00\x00\t\x00\x00\x00\x05\x00\x00\x00\x11\x00\x00\x00%\x00\x00\x00\x05\x00\x00\x00\xf9\xab\x02\x00', '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0 [... cut ...]
However, it seems that the BOOL and DINT arrays of the UDT are not transferred. I can see this by comparing the read of two UDTs,

MyUDT = REAL[200], BOOL[224], DINT[7]

MyRealUDT = REAL[200]

The timer example implies that a UDT containing different data types can be transferred.

Any thoughts on the above, is this possible with the library presently? Perhaps I am missing something. Regrettably I am not a software professional so please excuse any glaring oversight on my part.

Thanks,

Include versions to

- pylogix:0.7.3
- python:2.7.16
- OS:win10

dmroeder, 2020-09-05:
Thanks for all of the details, I'm a little busy at the moment so it'll be a little bit before I can really dig into this.  Could I convince you to try a previous version, like 0.7.0 or 0.6.7 to compare in the meantime?

OwenDavid, 2020-09-05:
Hello, yes, sure, I will try both.

Since posting the above I doubled my efforts to get this working.

Thankfully, there are enough clues in the open and closed issues that lead me to be able parse my UDT data!

All of the data was packed into the first of the returned data fields, 

So using ret.Value[0]

ret.Value[0:2] - 2 bytes used for data type

retValue[2:802] - 200 REAL32 tags = 4 bytes*200 = 800 bytes

retValue[802:830] - 224 BOOL tags = 1 bit per bool, 8 bools per byte, thus 224 bools for 28 bytes

retValue[830:858] - 7 Dints = 4 bytes per DINT 8 7 = 28 bytes.

dmroeder, 2020-09-05:
So you were actually getting all of the correct data, it just wasn't in the order you were originally expecting?

OwenDavid, 2020-09-05:
> Thanks for all of the details, I'm a little busy at the moment so it'll be a little bit before I can really dig into this. Could I convince you to try a previous version, like 0.7.0 or 0.6.7 to compare in the meantime?

Reverting back to 0.6.7 on python 2.7.16 was not successful.

`from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = '192.168.37.1'
    ret = comm.Read('historian_data')

print len(ret.Value)
print ret.Value

>>>2
>>>U=`

So now we are reading only the first two bytes of an UDT of 858bytes.

Reverting back to 0.7.0 on python 2.7.16 was not successful.

Here we have similar behaviour to 0.7.3 with ret.value containing 9 data fields,

9
['U=P\x8d\x17>\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00P\x8d\x17>\x05\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\xe4\x07\x00\x00\t\x00\x00\x00\x05\x00\x00\x00\x14\x00\x00\x00\x15\x00\x00\x00\x1f\x00\x00\x00e@\x01\x00', '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0 [... cut ...] x00\x00\x00\x00\x00P\x8d\x17>\x05\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\xe4\x07\x00\x00\t\x00\x00\x00\x05\x00\x00\x00\x14\x00\x00\x00\x15\x00\x00\x00\x1f\x00\x00\x00e@\x01\x00']

If we look at the length of the replies,

for value in ret.Value:
    print len(value)

858
770
682
594
506
418
330
242
154

They decrease by a fixed amount of 88 bytes each time and seem to be truncated from the front. So last bytes are all the same.

OwenDavid, 2020-09-05:
> So you were actually getting all of the correct data, it just wasn't in the order you were originally expecting?

Yes! Thanks,

However, to summarise,

ret.Value is now a list of nine byte strings.

The first ret.Value[0] is the correct length and contains all the UDT data!

Then as we iterate through we see that the byte string becomes shorter by 88 bytes each time.

Comparing the last bytes of each string we can see that the first 88 bytes are being lost on each repeat.

Hope this is clear now.

OwenDavid, 2020-09-05:
[image]

Here is the wireshark trace - we can see the PLC responds with two pakets of 964 bytes

OwenDavid, 2020-09-05:
Hi,

Well... I think I fixed this issue by adding,

break

to line 1252 of eip.py in the _getReplyValues function

So...

    def _getReplyValues(self, tag_name, elements, data):
        """
        Gather up all the values in the reply/replies
        """
        tag, base_tag, index = _parseTagName(tag_name, 0)
        data_type = self.KnownTags[base_tag][0]
        fmt = self.CIPTypes[data_type][2]
        vals = []

        data_size = self.CIPTypes[data_type][0]
        numbytes = len(data)-data_size
        counter = 0
        self.Offset = 0

        while True:
            index = 2+(counter*data_size)
            if index > numbytes:
               break
            if data_type == 160:
                tmp = unpack_from('<h', data, 2)[0]
                print tmp, len(data), vals
                if tmp == self.StructIdentifier:
                    # gotta handle strings a little different
                    index = 4+(counter*data_size)
                    name_len = unpack_from('<L', data, index)[0]
                    s = data[index+4:index+4+name_len]
                    vals.append(str(s.decode(self.StringEncoding)))
                else:
                    d = data[index:index+len(data)]
                    vals.append(d)
    
becomes....

    def _getReplyValues(self, tag_name, elements, data):
        """
        Gather up all the values in the reply/replies
        """
        tag, base_tag, index = _parseTagName(tag_name, 0)
        data_type = self.KnownTags[base_tag][0]
        fmt = self.CIPTypes[data_type][2]
        vals = []

        data_size = self.CIPTypes[data_type][0]
        numbytes = len(data)-data_size
        counter = 0
        self.Offset = 0

        while True:
            index = 2+(counter*data_size)
            if index > numbytes:
               break
            if data_type == 160:
                tmp = unpack_from('<h', data, 2)[0]
                print tmp, len(data), vals
                if tmp == self.StructIdentifier:
                    # gotta handle strings a little different
                    index = 4+(counter*data_size)
                    name_len = unpack_from('<L', data, index)[0]
                    s = data[index+4:index+4+name_len]
                    vals.append(str(s.decode(self.StringEncoding)))
                else:
                    d = data[index:index+len(data)]
                    vals.append(d)
                    break

Now reading a UDT returns only one ret.Value field which contains the full field, which seems to be correct.

Hope this helps!

OwenDavid, 2020-09-05:
Hi again,

Also, if we change eip.py, line 66,

from

self.CIPTypes = {0: (0, "UNKNOWN", '?'),
                         160: (88, "STRUCT", '<B'),

to

self.CIPTypes = {0: (0, "UNKNOWN", '?'),
                         160: (1, "STRUCT", '<B'),

Then we can read small UDTs.

This wasn't working when I was trying to troubleshoot earlier as the ret.Value field would always return None.

dmroeder, 2020-09-09:
Okay, I got a little time to look at this.  There were a few things going on.  I broke reading data larger than a single packet (unrelated to reading a struct).  Also trying to follow my timer example was broken, None was returned rather than than the raw bytes.

I never originally intended to return the raw data when reading a struct.   I noticed that the values would be returned, so at some point I decided to just return those values in case users wanted to parse the data themselves.  It has been a headache.

So I think the data is returned properly now, with the exception that if the reply is larger than a single packet.  That still won't work properly.  That will take a bit more work to sort out so I'll leave this open.  Try it out and let me know what you think when you get time.

OwenDavid, 2020-09-09:
Ok, I will try the updated version today.

I'd argue that the ability to correctly parse the raw bytes of UDTs is the most performant way to transfer large amounts of data periodically. So thanks for sticking with it!

In my recent experience the performance difference between reading a list of tags, versus reading several arrays in separate comm.Read messages, versus reading one UDT that wrapped the same arrays has been phenomenal. It has reduced our network traffic and the PLC comms overhead and will likely allow us to poll the PLC at 20hz (every 50ms) whereas we have been pushing the limits at 10hz beforehand.

I'd be pleased to see this developed and am happy to test and offer feedback. Thanks!

dmroeder, 2020-09-09:
No argument necessary, I fully understand that fact.  Raw performance wasn't one of my goals, although the faster the better.  I've tried my best to focus  on making something easy to use.

Anyway, I think I have a solution, I need to spend some time testing though to make sure I didn't break something else.  I need to rewrite my tests, a few thing slipped through that shouldn't have.

OwenDavid, 2020-09-09:
Yes, my progress has been testament to that! Being able to access a tag value in three or four lines of code using pylogix catalysed a review that led to a complete overhaul in our data transfer protocol. We may even be able to move away from using the kepware OPC server as a result! It would likely have been overlooked had it not been for the real ease of use!

So I ran the updated eip.py in 0.7.4 and it broke the working solution I had copy and pasted above. (I don't propose this as robust solution - more like a proof of concept)

From the Allen Bradley Literature:

The Tag Type Service Parameter for structures is a 4-byte sequence. The first two
bytes are the values A0 and 02, followed by the latter two bytes, which contain a
16-bit calculated field called a Structure Handle. When transmitted on the wire, it
looks like this:
A0 | 02 | Structure Handle Low Byte | Structure Handle High Byte

Can we check for these first two bytes and then return the raw bytes from position [4:].

I think this would satisfy the UDT crowd, and allow the rest of the struct code to work with the other cases.

dmroeder, 2020-09-09:
The changes I'm testing return the data starting at [4:].  I think I was originally returning data at [2:].

dmroeder, 2020-09-10:
Try the latest commit if you get a moment.

dmroeder, 2020-10-01:
This appears to be fixed to me, if you disagree @OwenDavid you can open this issue again.

closed 2020-10-01 by dmroeder