dmroeder/pylogix #143: unknown tag: __CONTAINER

AndrewStuhr, 2020-09-10:
Hello

I'm reading the tag list from a 5069-L3100ERM/A.  One of the tags returned is called "__CONTAINER".  It is type DINT.  When I read it's value, pylogix returns 0.  Thisi tag wasn't added by the programmer; in fact the programmer doesn't know what it is.

Do you know about this tag?  It seems like an internal tag to the PLC.  In general, do you know if tags that begin with 2 underscores are internal tags?

ottowayi, 2020-09-10:
That's a system tag that isn't being filtered out.
[image]

AndrewStuhr, 2020-09-10:
You're the man ottowayi.

So is this something that pylogix should filter, or I as the user should filter?

Thanks!

dmroeder, 2020-09-10:
Oh thanks @ottowayi  and @AndrewStuhr , I'll dig into that.

ottowayi, 2020-09-11:
The only rule I skipped was the I kept ones with colons in the name, because I wanted (well, actually someone opened an issue saying they were missing) module-defined tags in the list.

dmroeder, 2020-09-11:
That should be filtered out now @AndrewStuhr, as well as the AOI (UDI:) from issue #128

closed 2020-10-01 by dmroeder