dmroeder/pylogix #144: struct.error: unpack_from requires a buffer argument

cpapplefamily, 2020-09-12:
I have been testing the examples on a fresh os raspberry pi. I have not been able to get the example 40_read_timer.py to work. The error I continue to get is as follows

Traceback (most recent call last);
   File "40_read_timer.py", line 41, in <module>
      t = Timer(ret.Value)
   File "40_read_timer.py", line 21, in __init__
      self.PRE = unpack_from ("<i", data, 6)[0]
struct.error: unpack_from requires a buffer argument

When I try to read the 'TimerTest.ACC' using the following and Print I get results so the tag exists. 

ret = comm.Read('TimerTest.ACC") 
print(ret.value)   # returns current time

dmroeder, 2020-09-12:
Sorry about that.  Some recent changes broke the ability to read structs.  I have fixed it since, I updated the example a few minutes ago.  Try downloading the latest code from github and look at the latest example.

dmroeder, 2020-09-12:
Wait, this is a little confusing.  While I did have an issue with my timer example, your post has some things that aren't in my exmaple:

`ret = comm.Read("TestTimer.ACC")`

is not from my 40_read_timer example.  Neither is:

`print(ret.value) # returns current time`

cpapplefamily, 2020-09-12:
Yes sorry for the confusion. The test lines I created where to prove the tag existed in the PLC.

cpapplefamily, 2020-09-12:
I seem to still be getting the same error. 
I have not fully downloaded the full repository but used the `pip install pylogix` then copied each example as I wished to review them in a new file.py. Using the `pylogix.__version__` returns 0.7.0

[image]

kodaman2, 2020-09-12:
`pip install` grabs pylogix from pypi which is 0.7.0, you need to download the repository then follow the readme on how to install it from the repo. 

pypi isn't updated automatically when the repo is updated.

dmroeder, 2020-09-14:
> Yes sorry for the confusion. The test lines I created where to proof the tag existed in the PLC.

Nah it's my bad, I re-read your post, it was clear.  As @kodaman2 said, I haven't updated pypi recently, so to test, you'd have to download the latest code from here to test.

dmroeder, 2020-10-01:
This should be resolved though you will want to refer to the updated example, the bytes don't align with the original example anymore.

closed 2020-10-01 by dmroeder