dmroeder/pylogix #145: ESP32 micropython support

ivancho523, 2020-09-21:
Hello
How can I use this library with an ESP32, what changes should I make to be able to use it in a microcontroller? is it possible to use it with micropython?

kodaman2, 2020-09-21:
Hi, we don't explicitly support micropython. I am not even sure if micropython has all the core modules that pylogix needs like socket, etc.

Also you need ethernet for pylogix to communicate with the plc, wireless communication won't do.

By all means let us know if you do get it working. I just gave micropython a quick read and it does come with some core modules but prefixed with a `u` see below. So you will have to change those modules like `socket` to `usocket` if there's a module needed and not on micropython you'll have to learn how to compile and optimize that module for micropython. 

http://docs.micropython.org/en/latest/library/index.html

Best thing I can say is just to buy a linux based microcontroller that has raspbian like raspberry pi or zero or something alike there are several clones. This way you save a lot of time and frustration.

kodaman2, 2020-09-26:
Closing ticket, if you need any further assistance let us know.

closed 2020-09-26 by kodaman2

TheFern2, 2023-10-17:
@ivancho523 probably 3 years too late but nevertheless better late than never. We officially support micropython 1.20+. Check out the documentation for more information.