dmroeder/pylogix #147: How do you specify what local network resource is used to connect to a PLC?

10062244, 2020-10-23:
Hello,
I have an application where I need to connect to two machines that have the same IP address. My plan was to use a raspberrypi and two usb-ethernet adapters to connect to the two machines and the main ethernet port of the pi to connect to a data collection/distribution plc. I would like to use pylogix to move data between all three plc's connected to three different network interfaces. Is this possible with pylogix? Can you specify what local network resource you want to use when connecting to a PLC?

Thanks

dmroeder, 2020-10-24:
The short answer is no.

Might be possible in the future though, I've never encountered a situation where I had to do this.  Looks like the socket module has an option SO_BINDTODEVICE, which might allow this.  I'll have to do some homework, I don't immediately have the right hardware at my disposal to test this.

TheFern2, 2020-10-24:
Talk to the IT guy and slap him across the head. Most setups have a controls network with one or two subnets depending on how many devices you have, but to have multiple identical subnets seems like asking for trouble. Usually LANs are okay if they will always be independent, up until you want to communicate like you want to do now. 

I'd recommend looking into ip routes, in theory it should work, but don't quote me on it, I'm not a networking guy. 

From my basic understanding of routes you setup a destination ip in this case your plc identical ip, and a gateway ip for the usb-ethernet. You then use pylogix to communicate through this gateway and the route should send packets back and forth between the gateway and the plc.

I'll see if I can test my theory if I have some time this weekend. 
  https://www.techrepublic.com/article/understand-the-basics-of-linux-routing/

The other option would be to use a web server. Put a raspberry on each plc to pull your data, and send it to a server. Below was a quick example I wrote a while ago for simple rest requests, so you could run that simple server on each identical plc raspberry pi and the server ip needs to be different. And then you can request data from that server by doing simple rest requests, but you'll need one raspberry pi on each plc + one to converge your data. 

https://github.com/TheFern2/pylogix-api

evaldes2015, 2020-10-24:
I think you can do what you want if you use a couple of Rockwell Stratix switches: 
https://literature.rockwellautomation.com/idc/groups/literature/documents/qs/iasimp-qs038_-en-p.pdf

dmroeder, 2020-10-24:
@evaldes2015, yeah this sort of thing is typically done with hardware as you suggest.  We often supply multiple machines with the same addresses, each with a 1783-NATR or something like that to get them on the facility network.

TheFern2, 2020-10-24:
Nice solutions. I've never seen this setup, but I guess is more for identical machines on production lines which I haven't worked with before.

[image]

I think it might be cheaper with the NATR module, than buying 2 stratix switches + a layer 3 switch if you don't already have one.

10062244, 2020-10-24:
Hey guys,
Thanks for the ideas. I think what my plan is, is to connect a pi to each plc and network those with spi or i2c. Then exchange data between the three over that network. Doing something like that will probably be less then a third of the price of one NATR. I have our department setup with every cell behind a NATR now. Every cell has one of the machines I was referring to in my previous post now (which is a small packaging machine). This app is adding a second one of those to the cell to increase "time to operator intervention" lol. Just wanted to be able to have a setup roll one up to a cell without have to change ip address's. Worst case scenario I have to hardwire them together, but I would like to keep the all the same and I don't want to have to re-wire them all.

TheFern2, 2020-10-25:
Pi's are cheap indeed and very powerful too. If you're going that route a pi on each plc, you can just add another usb ethernet adapter to each pi so for example if your machine is on the 192.168.1.10, the pi eth0 can be 192.168.1.50, and your usb ethernet eth1 on 10.10.10.50. Your second pi on 10.51, 1.51... and so on.

Then add a switch so that your main data retrieval machine can communicate with all the 10.10.10.X devices. Or even another pi, if is only two plc's.

If you take a look at the simple flask server I posted earlier you can then setup the each server on eth1 for each plc. I've only created 3 routes, but if you need more, PRs are welcomed. I think retrieving data this way trumps spi/i2c all day every day. 😃 

https://github.com/TheFern2/pylogix-api
https://youtu.be/JIagCipFybE  <- Demo of the web api

evaldes2015, 2020-10-25:
Maybe you should try doing the port forwarding in Linux.
https://www.systutorials.com/port-forwarding-using-iptables/

TheFern2, 2020-10-25:
@evaldes2015 Yup that would be the easiest thing to do with minimal hardware. If routes are too difficult or just don't work then next option would be the web api. 

Interested to see if routes works though. We usually have a bunch of routes on our stations but I've never dug too deep into how they work together with stratix.