dmroeder/pylogix #148: Read function may have bug when read ["PC_TEMPZONE[0]"、"PC_TEMPZONE[1]"]

su600, 2020-11-11:
Hi !
I have a taglist like this:  ["PC_TEMPZONE[0]"、"PC_TEMPZONE[1]"、"PC_TEMPZONE[2]"], which PC_TEMPZONE is an array of REAL .

By using pylogix==0.7.5, the return value of this taglist is all about PC_TEMPZONE[0], I can't get the value of PC_TEMPZONE[1] and PC_TEMPZONE[2], maybe the index in tagname is not work.

When I change the pylogix version from 0.7.5 to 0.6.4, this problem doesn't exist. 
Maybe it is a Bug?

TheFern2, 2020-11-11:
Hi, 

Can you please show your code. So you're saying the same code works fine in 0.6.4 but not latest? Sounds like a possible bug.

Can you do me a favor and run example 4, but instead of 500 just however many indices you have.

What does that return?

su600, 2020-11-11:
> Hi,
> 
> Can you please show your code. So you're saying the same code works fine in 0.6.4 but not latest? Sounds like a possible bug.
> 
> Can you do me a favor and run example 4, but instead of 500 just however many indices you have.
> 
> What does that return?

I will try it later. 
I found #136 said the same problem,  he also said：
```
where array1 is SINT[100] and array2 is SINT[100]
Now if I read the tag list in one attempt, it returns only the first index of each array:
```

TheFern2, 2020-11-11:
Ah I see if you're passing the list to the read function then yes it shouldn't work, but is interesting you said it work on 0.6.4.

Example 4, should work fine for your case. Or use a for loop to pass one tag at a time to the read function.

@dmroeder did read used to be able get a list of tags? I don't recall that functionality.

dmroeder, 2020-11-11:
@TheFern2 I believe when @su600 is saying taglist, his referring to the list of tags passed to Read, not the list of tags returned from GetTagList().

This is a bug, I'll get it fixed soon.

dmroeder, 2020-11-11:
This should be fixed now.

Regarding issue #136, the way I understood that one is that he was trying to read multiple array elements in a list, rather than them individually.  Though now that I see the issue you raised, he would have been having this problem as well.  I didn't get any clarification on that issue...

TheFern2, 2020-11-11:
> @TheFern2 I believe when @su600 is saying taglist, his referring to the list of tags passed to Read, not the list of tags returned from GetTagList().
> 
> This is a bug, I'll get it fixed soon.

Got it, I misunderstood the issue. That's why is always better to see the code. I understand now what the issue was.

su600, 2020-11-12:
> Regarding issue #136, the way I understood that one is that he was trying to **read multiple array elements in a list**, rather than them individually.

Yes, sometimes my taglist is a list made of **array elements, which may not continually**, so I put the elements I need into a list.

closed 2021-01-07 by su600