dmroeder/pylogix #149: Can pylogix read Micro800 program Tags?

10062244, 2020-11-18:
Hello,
Ive been trying to read some program tags out of a Micro800 program with no luck. I can read read controller scope tags fine. Is there a change in the syntax when reading from a Micro800 program vs a controllogix program? I did the Program:programName.tagName scheme. Also, the GetProgramsList function doesnt seem to work either for my Micro. 

Thanks

TheFern2, 2020-11-18:
Hi, you can only read global variables (controller scope) on micro800.

https://github.com/dmroeder/pylogix/issues/66#issuecomment-533271724

dmroeder, 2020-11-22:
Rockwell doesn't provide a documented way to read local variables.  Also, the Micro8xx series doesn't support a lot of the services that the Logix platform supports.  These features only work with the Logix platform, it's a limitation of the Micro8xx.

closed 2020-12-09 by TheFern2