dmroeder/pylogix #150: Micro850 not showing up on RPI4

dpatel2016, 2020-12-09:
I am trying to run the discover.py example with Micro850 connected to RPI 4. The program runs but does not print anything. is there any ethernet settings that need to changed?

TheFern2, 2020-12-09:
Not entirely sure Discover will work with Micro PLCs I think that is meant for Controllogix PLCs, make sure you can ping the plc from the pi. Then try https://github.com/dmroeder/pylogix/blob/master/examples/01_read_simple.py change the tag to one of your tags, and your PLC ip. On Micro you can only read global tags not program tags so that's something to be aware.

dpatel2016, 2020-12-09:
I tried to read it using read_loop example, but it just prints "None".

TheFern2, 2020-12-09:
> I tried to read it using read_loop example, but it just prints "None".

Ah sorry please put this after the IP:

```
comm.Micro800 = True
```

dpatel2016, 2020-12-09:
I added that the first time, but still prints "None"
I think it has to do with RPI4 ethernet settings.

TheFern2, 2020-12-09:
Are you able to ping the PLC?
What is your pylogix version? Can you do `pip list`, and see what is your pylogix version.

dpatel2016, 2020-12-09:
I figured it out! It was changing the raspberry pi ethernet to static IP address.

closed 2020-12-09 by TheFern2