dmroeder/pylogix #155: Batch reading NoneType AttributeError: 'NoneType' object has no attribute 'group'

tdesroches, 2021-01-08:
### Preflight checks

- [x] Have you tried the examples?
- [ ] Have you tried [pylogix-tester](https://github.com/kodaman2/pylogix-tester)?
- [x] Have you read previous issues?

## Type of issue

- [x] Bug
- [ ] Feature Request
- [ ] Question
- [ ] Other

> Delete items that do not apply below.

## Description of issue
When reading a batch of variables the batch read fails when a BOOL is included.  The first 4 variables in the list are REALS the 5th one is a BOOL.  If I read the BOOL alone it works, if I read the bool along with any of the others it fails.

## Expected behavior
Expected response with tags and values

## Actual behavior
Reading generates Attribute Error
NoneType AttributeError: 'NoneType' object has no attribute 'group'

## Code
```
from pylogix import PLC
import time

with PLC() as comm:
    comm.IPAddress = '192.168.1.221'
    tags = comm.GetTagList()
    print(tags.Status)
    if tags.Status == 'Success':
        iris_tags = [ t.TagName for t in tags.Value if 'IRIS' in t.TagName]
        #comm.SetPLCTime()
        plctime = comm.GetPLCTime()
        print(plctime.Value)
        print(iris_tags[4])
        read = True
        while read:
            try:
                ret = comm.Read(iris_tags[:5])
                #print(ret.Value)
                for r in ret:
                    print(r)
                    print()
                time.sleep(1)
            except KeyboardInterrupt:
                print('exiting')
                read = False
```

## Screenshots
[image]

## Stacktrace

## Versions

Include versions to

- pylogix: 7.5
- python: 3.8.5
- OS: Ubuntu 20.04

dmroeder, 2021-01-08:
Can you try the latest code from the repo?

TheFern2, 2021-01-09:
Can you show what iris_tags list contains, remember that GetTagList contains program names and other items besides just tags.

dmroeder, 2021-01-09:
The first 4 tags were a REAL, then a BOOL.  I think this is the same issue I fixed the other day.

TheFern2, 2021-01-09:
The bool strikes again haha. Yeah you're right that should take care of it.

dmroeder, 2021-01-09:
If it is true in this case, I'll get an update to pypi.

tdesroches, 2021-01-09:
I'm back in the office on Monday, I'll try the repo version and get back to you guys.  Thanks

tdesroches, 2021-01-11:
Ok so I did some testing, I found the variable I thought was a BOOL is actually a BOOL Array, my apologies.  If I include the BOOL by selecting a single index it works no problem.  Is this a bug or is it just that arrays need to be handled seperately?

tdesroches, 2021-01-11:
I figured out how to get the DataType, however the BOOL Array DataType is DWORD, is that expected?

tdesroches, 2021-01-11:
Ok I think this issue can be closed.  For anyone else there apparently is no such thing as a BOOL Array in AB.  When declaring a BOOL Array in the controller they can only be in increments of 32 BOOL because it seems internally they are arrays of DWORDs.  The Array length returned is the number or these DWORDs in the Array.  ie a Bool Array of 64 BOOL is a DWORD array or length 2.

TheFern2, 2021-01-11:
> Ok so I did some testing, I found the variable I thought was a BOOL is actually a BOOL Array, my apologies. If I include the BOOL by selecting a single index it works no problem. Is this a bug or is it just that arrays need to be handled seperately?
> Is there a way to get the tag type when I'm reading the tag?

Before we added the Response object, we used to return LgxTag obj now Tag class and that object had DataType.

[image]

[image]

Am I reading this right @dmroeder? It looks like we're not returning this for the user.

Tag class has a __str__ method but we're not currently returning the Tag object for the user to use it.

dmroeder, 2021-01-11:
@TheFern2 The LgxTag class is only used for GetTagList().

BOOL arrays are the worst.  As you discovered @tdesroches, they are not stored in the PLC as BOOL, but 32 bit words.  That is the reason why when you create BOOL arrays, you have to create them blocks of 32.  You can't create a BOOL array of 16 for example.  When I request BOOL[4] for example, I have to send the request with an index of 0, because bool 4 is in the first DWORD.  Think of BOOL arrays as an array of DINT's and you are using the individual bits.

A few days ago, I realized that I was not handling the index properly for some time.  I believe this to be fix in the latest code here on GH, but it has not been put up on pypi.  Try the latest code here, the problem should be resolved.

tdesroches, 2021-01-11:
@dmroeder the tests today were performed with the code from gh and it didn't appear to have any problems reading a single index,  I haven't tried the pipy version though.

closed 2021-01-11 by tdesroches

dmroeder, 2021-01-11:
The version on pypi is the same version you found the issue on.  I'll get pypi updated tonight.

TheFern2, 2021-01-11:
> The version on pypi is the same version you found the issue on. I'll get pypi updated tonight.

Btw I meant to mention this last time, you can add a batch file or bash file with the deployment pypi command to the repo.
And on your machine you can have a .pypirc with your credentials. Just to make it a little easier.

https://dev.to/thefern/package-and-deployment-jh
https://github.com/TheFern2/OneDriveIgnore On this repo I have a build and deploy script files.

kyle-github, 2021-01-11:
Thanks for the discussion on BOOL arrays.  I almost never use them in PLC programming so I never did much with them in my library, but...  I think I need to as this is clearly painful :frowning_face: