dmroeder/pylogix #159: Boolean Arrays maxed at 3968 index

TheFern2, 2021-02-12:
## Type of issue

- [ x ] Question

## Description of issue

I was testing boolean arrays just like we have a HugeArray in the  pylogix ACD test program, but pylogix only retrieves 3968 indices for a boolean array, in my case I had BOOL[10000]. Not sure if this is intended due to how booleans are, I did retest DINT arrays and those are fine I tested with 10000, and even higher up to 60000.

## Expected behavior

Read the entire array.

## Actual behavior

Maxed at index 3968

## Versions

Include versions to

- pylogix: 0.7.8
- python: 3.8.5
- OS: Ubuntu Server 20.04

dmroeder, 2021-02-12:
Boolean arrays make me want to quit.

TheFern2, 2021-02-12:
> Boolean arrays make me want to quit.

Lmao I doubt someone is using boolean arrays of >4000, right? So maybe is a feature. I was just doing a quick test how fast I could read/write data from plc to redis, and was using large arrays to test the delay. I really just stumbled across that, is a non issue for me.

dmroeder, 2021-02-12:
If I were to guess, ~4000 is about what fits in a single packet so I'm probably not getting the values from the subsequent packets.  If you were to specify a ConnectionSize smaller, say 508, you will get fewer values returned.

I'll get to the bottom of it.  Bool arrays are my nemesis.

kyle-github, 2021-02-12:
BOOL arrays are rounded up to the nearest 32-bit segment and packed into DINTs.   So if you have an array of 35 BOOLs, you'll use two DINTs.   You will get back 8 bytes from the PLC.

So 3968 is exactly 124 DINTs.   That's a little suspicious.   That happens to be about the maximum number of DINTs you can have in a 500+ bytes packet.

I very much second Dustin's complaint.   Someone was off their meds when they came up with how Rockwell deals with BOOL arrays.

"Oh, you think UDT field packing is weird?  Hold my beer!"

dmroeder, 2021-02-12:
Indeed @kyle-github.  I'm onto what is happening, it has to do with how I'm sending the request for the next chunk of data.

dmroeder, 2021-02-13:
Latest commit should take care of it

TheFern2, 2021-02-13:
> Latest commit should take care of it

Nice work!

closed 2021-02-13 by TheFern2