dmroeder/pylogix #163: Write to Micro850

dpatel2016, 2021-03-08:
It's not writing to the PLC consistently. Sometimes it works, and sometimes you need to send the signal twice to write it. How do I fix this? I am using Micro850 and the variable is Boolean. I am only writing 0 or 1.

closed 2021-03-09 by dpatel2016

dmroeder, 2021-03-09:
I see you closed it, could I ask what was happening?

dpatel2016, 2021-03-09:
I have micrologix and micro850 PLCs so I mainly use node red and mqtt to python to control M850s. In the python code, I was subscribing and based on the msg, writing to M850. It was not working because the pylogix code was outside the on_connect part of mqtt. I had to press button twice on node red to write it not sure why, but after moving the pylogix code to inside mqtt function, it started working as expected.

dmroeder, 2021-03-09:
Okay, thanks for the details.