dmroeder/pylogix #164: Pylogix as a part of a simple Web Project

GitHubDragonFly, 2021-03-10:
This is only intended to show a possible way of using the pylogix library to read tags and show their values in a web page:

https://github.com/GitHubDragonFly/WebProject

Close this non-issue whenever you want.

TheFern2, 2021-03-10:
@GitHubDragonFly That is very cool, I had built a very simple similar API a while back in flask though. 
https://github.com/TheFern2/pylogix-api

@dmroeder I think it might be a good idea to add a doc with all these cool projects built with pylogix.

GitHubDragonFly, 2021-03-11:
I like your API, since it's somewhat simpler.

If you would like some improvements here is the app.py file that allows reading array of tags as well as elements of 1-dimensional array, addressed like this:

localhost:5000/pylogix/v1.0/plc/192.168.0.10/0/tags/[someTag1, someTag2, someTag3, someTag4]
localhost:5000/pylogix/v1.0/plc/192.168.0.10/0/tags/someArrayTag[x]{y}

Both x and y need to be within the array bounds and also y > 0.

If you like it then just include it in your repo and spare me from submitting a PR.

[app.zip](https://github.com/dmroeder/pylogix/files/6120316/app.zip)

Your idea about creating a doc or a text file with all the cool project links sounds good to me and would poke people in the eye if it is a part of "examples".

I will eventually go ahead and close this non-issue.

GitHubDragonFly, 2021-03-11:
There was one typo so I just replaced the file in my previous post.

TheFern2, 2021-03-11:
@GitHubDragonFly Thanks for your contribution, I've added you to that commit for proper credit.

closed 2021-03-11 by TheFern2

GitHubDragonFly, 2021-03-12:
If you ever go about creating a list of pylogix projects, here is another one that was done couple of years ago and does require some understanding:

    https://github.com/tdw1004/PylogixGUI

GitHubDragonFly, 2021-03-21:
If you might ever be interested in running this web project in Linux, I did bother converting the batch file to shell script.

It does need to be first made executable by running chmod +x on it.

GitHubDragonFly, 2021-03-29:
Here are some pictures of how pylogix can also be used with Node-RED (this was done on Raspberry Pi 2).

It is a modified version of this example:  https://flows.nodered.org/flow/778859ca2503db35ff0e12341508efef

Some benefits of Node-RED would be already available GUI elements and other packages (like pythonshell used in this example), minimal python code, cyclic reading and live dashboard updates.

[image]

[image]

GitHubDragonFly, 2021-04-01:
There is a new repo for this Node-RED example project:  https://github.com/GitHubDragonFly/Node-RED_PLC_Example

It is a hybrid project, using pylogix library for tag reading and using pycomm3 library for the device discovery and tag listing.