dmroeder/pylogix #173: Multiwrite Issue

alexkireeff, 2021-05-03:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [x] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [x] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [x] Have you read previous issues?

## Type of issue

- [x] Bug
- [ ] Feature Request
- [ ] Question
- [ ] Other

## Description of issue
Code example explains it better than I can, but basically, calling comm.Write with the syntax for multiple tags when you have one tag causes it to not write anything to the tag.
Expected behavior would be comm.Write writing the correct value to the tag.

## Code
`from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = ''

    comm.Write('TEST_DINT', 0)
    print(comm.Read('TEST_DINT').Value)
    # expected: 0
    # actual: 0

    r = comm.Write([('TEST_DINT', 1)])
    print(comm.Read('TEST_DINT').Value)
    print(r[0].Status) # Path segment error
    # expected: 1
    # actual: 0

    comm.Write([('TEST_DINT', 1), ('NOT A TAG', 1)])
    print(comm.Read('TEST_DINT').Value)
    # expected: 1
    # actual: 1`

## Versions

- pylogix: 0.7.8
- python: 3.7.3
- OS: Debian 10.9

dmroeder, 2021-05-03:
I see, so it appears that the issue is if you pass a list of 1 write, things go sideways.  I'll sort this out.

dmroeder, 2021-05-03:
Should be fixed with the latest commit (0.7.12).  Thanks for bringing it to my attention.

alexkireeff, 2021-05-03:
Awesome, thanks!
I think you might have meant to return a list on line 118?
https://github.com/dmroeder/pylogix/blob/master/pylogix/eip.py#L118

dmroeder, 2021-05-03:
Ahh shoot, you are correct.  I need to get it together :)

alexkireeff, 2021-05-03:
Haha, it's all good. Thanks again!

closed 2021-05-03 by alexkireeff