dmroeder/pylogix #174: multiwrite writes to incorrect bits

alexkireeff, 2021-05-03:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [x] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [x] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [x] Have you read previous issues?

## Type of issue

- [x] Bug
- [ ] Feature Request
- [ ] Question
- [ ] Other

## Description of issue
When writing to specific bits in a multiwrite, pylogix writes the first bit instead of the specified bit.
Writing to a specific bit works fine when it is not a multiwrite.

## Code
```
with PLC() as comm:
  comm.IPAddress = '192.168.0.2'

  comm.Write([('TEST_DINT.3', 1), ('NOT A TAG', 1)])
  print(comm.Read('TEST_DINT').Value)
  # expected: 8
  # actual: 1

  comm.Write('TEST_DINT.0', 0)
  print(comm.Read('TEST_DINT').Value)
  # expected: 0
  # actual: 0
```

## Versions

- pylogix: 0.7.12
- python: 3.7.3
- OS: Debian 10.9

dmroeder, 2021-05-03:
Sigh, bits and boolean arrays are my nemesis.  :(

dmroeder, 2021-05-03:
@alexkireeff you mind giving the nemesis branch a whirl?

alexkireeff, 2021-05-03:
Haha, yeah they are my worst nightmare too.

I just ran with the branch, it seems to be working.

dmroeder, 2021-05-03:
Great, thanks.  I did test it, but after you immediately noticed a problem with my previous "fix", I thought I'd have you try it too :)

alexkireeff, 2021-05-03:
Lmao, no worries. Have a good one!

closed 2021-05-03 by alexkireeff