dmroeder/pylogix #175: multiwrite bug

alexkireeff, 2021-05-12:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [x] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [x] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [x] Have you read previous issues?

## Type of issue

- [x] Bug
- [ ] Feature Request
- [ ] Question
- [ ] Other

## Description of issue
When writing to specific bits in a multiwrite, pylogix writes the first bit instead of the specified bit.
Writing to a specific bit works fine when it is not a multiwrite.
Similar to [previous issue](https://github.com/dmroeder/pylogix/issues/174).

## Code
```
from pylogix import PLC

with PLC() as comm:
  comm.IPAddress = '10.10.100.2'

  comm.Write('TEST_DINT', 0)
  comm.Write('Test_float', 0)
  print(comm.Read('TEST_DINT').Value)
  # expected: 0
  # actual: 0
  print(comm.Read('Test_float').Value)
  # expected: 0.0
  # actual: 0.0

  r = comm.Write([('TEST_DINT.2', 1), ('Test_float', 1)])
  print(comm.Read('TEST_DINT').Value)
  # expected: 4
  # actual: 1
  print(comm.Read('Test_float').Value)
  # expected: 1.0
  # actual: 1.0

```
## Versions

- pylogix: 0.7.13
- python: 3.7.3
- OS: Debian GNU/Linux 10 (buster)

alexkireeff, 2021-05-12:
Here's a slightly clearer example:
```
from pylogix import PLC

with PLC() as comm:
  comm.IPAddress = '10.10.100.2'

  comm.Write('TEST_DINT', 0)
  comm.Write('Test_float', 0)
  print(comm.Read('TEST_DINT').Value)
  # expected: 0
  # actual: 0

  r = comm.Write([('TEST_DINT.2', 1), ('Test_float', 1)])
  print(comm.Read('TEST_DINT').Value)
  # expected: 4
  # actual: 1
```

dmroeder, 2021-05-12:
You sure @alexkireeff?  Have you tried with the latest code on the repo?

alexkireeff, 2021-05-12:
Yeah, I just tried to update my install with
```pip install -U pylogix```
and it said I was already up to date.
I also just ran the snippet on a PLC I have access to and got the following output:
[image]

GitHubDragonFly, 2021-05-12:
@dmroeder Just a suggestion to update your README.md installation steps with an additional single line using pip + git clone:

python -m pip install git+https://github.com/dmroeder/pylogix.git

This should either install the latest or update the currently installed pylogix version.

alexkireeff, 2021-05-12:
Yeah, I ran
```pip install git+https://github.com/dmroeder/pylogix.git```
and that fixed the bug.

dmroeder, 2021-05-12:
@GitHubDragonFly the readme used to say that before I put the project on pypi.  I changed the repo to mention cloning the repo and installing it rather than using pip, mainly because I figured that not everyone would have git installed.  Maybe I'll update the readme to be more clear, explaining that I only update pypi occasionally, so to try the bleeding edge either cloning the repo or installing with pip via git+path.

dmroeder, 2021-05-12:
So @alexkireeff, you opened this as an issue previously, I pushed the fix here to github, but did not yet update pypi (pip repo).

alexkireeff, 2021-05-12:
Ahhhh, so, sometime between now and when I last opened this issue, I 'updated' my pylogix install back to 0.7.13, which caused the issue to reappear.
Thanks :) and sorry about the confusion.

closed 2021-05-12 by alexkireeff