dmroeder/pylogix #178: Issues with writing tags

ShaliniBanerjee, 2021-07-02:
I have used Studio 5000 to program Logix 5571.

I could read the tags properly using pylogix. However, while writing the tags, even though the program is running successfully, the actual tag values are not getting modified.

Please suggest.

dmroeder, 2021-07-02:
What happens when you do the following:
```python
ret = comm.Write("MyTag")
print(ret.Status)
```

The most likely causes are that there is logic in the PLC writing a value to the tag (cross-reference the tag to see how it is used), the tag's access is set to read only or no external access.

ShaliniBanerjee, 2021-07-02:
Thank you for the advice. It is still not working.

I tried with the above code, it shows 'Success'.
I am trying to write a tag 'Start-Switch', which is used to manually start the program. It is not used in any other instructions. The tag also has  a Read/Write access.

TheFern2, 2021-07-02:
Hey, there, please post code.

ShaliniBanerjee, 2021-07-02:
This is the code:

from pylogix import PLC

comm = PLC()
comm.IPAddress = '192.168.1.24'
comm.ProcessorSlot = 0

result = comm.Write('Start_Switch',1)
print(result.Status)

comm.Close()

dmroeder, 2021-07-02:
I agree with @TheFern2, I'm sure it is not quite "Start-Switch", that isn't a valid tag name.  If the tag is aliased to an input, you won't be able to write to it.  Also, make sure you are online when you are judging if the write was successful.  Pylogix did write the value to the tag if it reported success.  So the only likely things at that point are that you either aren't online when viewing the tag database, the tag is addressed to an input or another instructing is controlling the value.

ShaliniBanerjee, 2021-07-02:
Yes, The tag is aliased to an input Local:2:I.Data.0.
Is there a way to write the value in the tag.

dmroeder, 2021-07-02:
If you inhibit the I/O module, then pylogix should be able to write to it.  Right now, the voltage on the input is deciding the state of the input.  Pylogix tries to write a 1, the fact that there is 0 volts on the input is writing it back to 0

TheFern2, 2021-07-02:
Ah yeah you're trying to override a hardware input. Here's an experiment for you add that input to a trend and see if there's a spike when pylogix tries to write 1. You'll probably need a really fast frequency, but I dunno if you would be able to see it on the trend. Either way you need a boolean tag that's not an input so pylogix can write to it.

dmroeder, 2021-07-02:
I should caution, only inhibit the module if this PLC is not in production.  Inhibiting it will disable all inputs on that module.

ShaliniBanerjee, 2021-07-02:
Thank you so much for the replies. I am still having some confusion.
The input and output tables are mapped to the hardware I/O modules. We should be able to write these tags as well, as monitoring systems should be able to override these tags if and when required. And these tags are most likely to be alias to some physical addresses. So the write() method does not take into account alias tags or tags referencing to physical addresses?

ShaliniBanerjee, 2021-07-02:
Right, I understand. I will try with the suggestions and see if it works.

evaldes2015, 2021-07-02:
I'm assuming you want to be able start you machine using pylogix.
Your best option here is not to try to write to this tag.
Modify your PLC program. Create another BOOL tag and put it in parallel with this "Start_Switch", then put an OTU at the end of the rung to unlatch the BOOL you wrote.

ShaliniBanerjee, 2021-07-07:
Thank you so much for all your replies.
It worked very much.

I can write in the alias tags(output) and it is getting reflected in the Image table of the controller as well as the output module.
For input alias tags, let us say, I want to start a switch( Start Switch: Alias to an input), I created a Boolean tag that is linked with this tag, so now, by writing to this Boolean tag, I can modify the value of the alias.

I have another question. When I am using Wireshark to dissect the packets with Read and Write requests for the I/O tags, it is showing, connected messaging. So the communication is not following Unconnected(Explicit) messaging? Is there a way to get Explicit requests and responses?

dmroeder, 2021-07-10:
An alias simply links 2 tags together. In your case, trying to write to the alias will give the same result because the input is controlling the state of the tag.

The connection is explicit, it is also using connected messaging. You have no control externally how pylogix accesses the PLC.

ShaliniBanerjee, 2021-07-21:
Thank you for the replies. That was really helpful.

closed 2021-07-21 by ShaliniBanerjee