dmroeder/pylogix #179: Accessing I/O tags

ShaliniBanerjee, 2021-07-21:
I have used Studio 5000 to program Logix 5571. I have used pylogix to read and write I/O tags. While using Wireshark to dissect the  packets, though I could see the services 0x4c and 0x4d, I could not find the class that is being accessed. I have to perform some filtering based on class and I am guessing that the Symbol Class is being accessed. Could you please give me some insight into this.

dmroeder, 2021-07-22:
I'd prefer emailing me or joining the discord for asking questions, rather than open issues.

Those services are Rockwell specific and largely undocumented.  This is the only documentation I have ever found:

https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf

ShaliniBanerjee, 2021-07-22:
Sure. Thank you.

closed 2021-07-22 by ShaliniBanerjee

dmroeder, 2021-07-22:
Let me know if that document helps at all

ShaliniBanerjee, 2021-07-22:
Hi,

Thank you for your reply.

I am working on doing some application level filtering with the I/O tags
based on Class it belongs to. I had read that I/O tags are exclusive to
Assembly Class and so was I keen on doing some filtering based on that. But
then it is mostly used for implicit messages. Since I am working with
explicit messaging, I figured out that the Symbol Class(0x64) might be
accessed. I have not found enough documentation about the Symbol Class
though. Anyhow, on dissecting the packets with Wireshark, I could not see
any class name mentioned. As my project deals with filtering the I/O tags
that will be communicated over an EtherNet/IP communication module, I
wanted to see what class the messages that are exchanged between the PLC
and the pylogix queries belong to.

I have already read the document. They have talked about the Symbol Class
in very vague detail as to why that class is being used. I wanted to ask
what class pylogix uses with the services 0x4c and 0x4d to read and write
the I/O tags and also, on a different note, if you could tell anything
about the Symbol Class.

Regards,
Shalini

ShaliniBanerjee, 2021-07-22:
Could I get your email id please

dmroeder, 2021-07-22:
Oh sure, it is in the setup.py file
dmroeder@gmail.com

I don't know any more than what was implemented in pylogix, which largely came from that document.  It has been a few years since I've had to pay attention to that portion of the protocol, I spent a lot of time in wireshark, scratching my head.

The reason that wireshark doesn't give you more details is because their CIP dissector only know what has been documented and these objects are not part of the CIP spec, they are "vendor specific" objects, at least that is how I understand it.