dmroeder/pylogix #184: Speed benchmarks

alex4acre, 2021-09-22:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [ ] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [ ] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [ ] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [x] Question
- [ ] Other

> Delete items that do not apply below.

## Description of issue
Has this been benchmarked in terms of reads/writes per second? What is a reasonable expectation? Can this be used to query data at a regular high interval (100 tags at 100ms?). 

## Expected behavior

## Actual behavior

## Code

Please provide a minimal, reproducible example

## Screenshots

## Stacktrace

## Versions

Include versions to

- pylogix:
- python:
- OS:

TheFern2, 2021-09-22:
Hey there, I did some performance testing before comparing this to OPC servers. A few hundred tags should be no problem at 100ms. Have a look at these reports. pylogix was used for the EIP portions. Hope this helps.

Avg response for 1000 individual tags (zero use of threading or async funcs), also this could be faster by a tenfold if you read an array of data:

```
1000 tags pylogix
No. of tags 1000, execution: 183.488 milliseconds
No. of tags 1000, execution: 70.272 milliseconds
No. of tags 1000, execution: 73.244 milliseconds
No. of tags 1000, execution: 67.356 milliseconds
No. of tags 1000, execution: 70.258 milliseconds
No. of tags 1000, execution: 68.312 milliseconds
No. of tags 1000, execution: 78.13000000000001 milliseconds
No. of tags 1000, execution: 82.938 milliseconds
No. of tags 1000, execution: 71.257 milliseconds
No. of tags 1000, execution: 75.163 milliseconds
```

[round-trip.pdf](https://github.com/dmroeder/pylogix/files/7210977/round-trip.pdf)

[large-read.pdf](https://github.com/dmroeder/pylogix/files/7210975/large-read.pdf)

dmroeder, 2021-09-22:
Dang @TheFern2, nice work.  It's kind of funny that Ignition beats RA at their own protocol.  I'd guess they utilize reads by instance ID, which allows you to pack more data into a packet, plus multiple connections.

alex4acre, 2021-09-22:
@TheFern2, are you saying that you are doing individual reads for each tag at this rate? That these are not put into a tag array like the example 05_read_multiple_tags.py?

TheFern2, 2021-09-22:
> @TheFern2, are you saying that you are doing individual reads for each tag at this rate? That these are not put into a tag array like the example 05_read_multiple_tags.py?

Correct single reads, it could be very well even faster if you group tags in a single call or do an array read. Not sure how many tags you can do per multiread with pylogix? @dmroeder is there a limit?

> Dang @TheFern2, nice work. It's kind of funny that Ignition beats RA at their own protocol. I'd guess they utilize reads by instance ID, which allows you to pack more data into a packet, plus multiple connections.

Yeah not sure what they're doing behind the scenes but I only ever saw one eip connection open in ignition. they're fast for sure. I am thinking RA is doing some internal throttling behind the scenes, ain't no telling.

dmroeder, 2021-09-22:
There is no limit.  pylogix will break the reads up based on the max packet size.

@TheFern2 what controller were you using for your test?

TheFern2, 2021-09-22:
> There is no limit. pylogix will break the reads up based on the max packet size.
> 
> @TheFern2 what controller were you using for your test?
> 
> Edit: based on a quick test, I believe @TheFern2 test would have been reading a list of tags, doing 1000 individual reads would take a lot longer than 100ms. I just did a test reading 300 tags, one at a time, it took about 3.4 seconds. Reading them as a list took about 70ms. Now if the controller is a 1756-L8x or a 5069-L3xER, the speeds will likely be much faster.

Yeah, that's right I had forgotten just checked my scripts. It was a list of tags indeed. It was a compactlogix L33 iirc.

alex4acre, 2021-09-22:
@dmroeder 
> There is no limit. pylogix will break the reads up based on the max packet size.
> 

from the documentation it seemed there was a byte limit. Are you saying that is no longer the case? Can I pack as many different tags into a single python array and read it that way?

dmroeder, 2021-09-22:
> @dmroeder
> 
> > There is no limit. pylogix will break the reads up based on the max packet size.
> 
> from the documentation it seemed there was a byte limit. Are you saying that is no longer the case? Can I pack as many different tags into a single python array and read it that way?

You should be able to read a list of tags as large as you want.

I might need to make the documentation a little more clear.  There is a limit to the amount of data that can fit into a single packet.  For modern Rockwell PLC's that limit is 4002 bytes.

Pylogix will take your list of tags and generate as many reads as necessary to read all of your tags, returning the values to you after all are read.

So let's say 100 tags will fit into a single 4002 byte packet.  You provide a list of 200 tags.  pylogix will generate 2 packets, the first 100 being in the first packet, the next 100 being in the 2nd packet.  Once all  of the values are retrieved, the enter list of 200 responses are returned to you.

Hopefully that makes sense.

alex4acre, 2021-09-22:
That makes sense!

closed 2022-09-10 by dmroeder

sillycrackers, 2025-06-02:
> > [@dmroeder](https://github.com/dmroeder)

> I might need to make the documentation a little more clear. There is a limit to the amount of data that can fit into a single packet. For modern Rockwell PLC's that limit is 4002 bytes.
>
> Pylogix [...]

I was also confused by the documentation. 

I avoided reading a list of tags in my project because I never knew if it would be over the byte limit. So I just stuck to reading tags individually. Based on what you are saying it's safe to read a list of tags even though it might go over the byte limit, because pylogix will handle it.

It would help others I think if you did clarify this in the documentation @dmroeder like you mentioned :)

Thankfully I found this, now I can improve the performance of my project.

dmroeder, 2025-06-02:
Great suggestion, I'll do that.

Originally you had to specify the packet size, then at some point I changed it to try the large packet size, then fall back to the small packet size.  Similarly, at one point it was up to the user to make sure that tags fit into a packet.  Eventually, pylogix was able to do this for you, I never addressed the documentation.  It should have occurred to me that if I had to explain it, then the documentation wasn't clear.